PX4/PX4-Autopilot #8538: ll40ls for rpi3

belaidH, 2017-12-30:
is the ll40ls compatible with the posix platform (rpi3) because when I compile I get error about (nuttx/clock.h and nuttx/wqueue.h).

dagar, 2017-12-30:
You'll need to go through and fix those issues. Remove the nuttx headers first, then readd generic headers as needed.

LorenzMeier, 2017-12-30:
Or in other words: It is reasonably easy and a pull request would be appreciated.

belaidH, 2017-12-30:
sorry I am not an expert. So I just have to remove the headers ? because after that several problems concerning the variable or function 'work' appear .

dagar, 2017-12-31:
@belaidH yes remove those headers, and in that case add `#include <px4_workqueue.h>`.

belaidH, 2017-12-31:
Okey thank you. I will try it

belaidH, 2018-01-01:
I think now it's okay for the workqueue. but I get an other error 
about the irqstate_t on the LidarLiteI2C.cpp file (I would use the I2C of the rpi). 

[image]

dagar, 2018-01-01:
@belaidH you can replace the critical sections with ATOMIC_ENTER/ATOMIC_LEAVE.

On NuttX it's the same thing - https://github.com/PX4/Firmware/blob/master/src/drivers/device/nuttx/cdev_platform.hpp#L10
On other platforms it's a lock - https://github.com/PX4/Firmware/blob/master/src/drivers/device/posix/cdev_platform.hpp#L6

belaidH, 2018-01-01:
@dagar thank you. after some tries it's compiled. Is there manip to start the ll40ls after launching the ./px4 because I don't see his topic when I use the "uorb top" command.

dagar, 2018-01-01:
Does it work if you start it manually `ll40ls start`?

Once you're sure it's working you could add it to the raspberry pi configs here. https://github.com/PX4/Firmware/tree/master/posix-configs/rpi

belaidH, 2018-01-02:
@dagar when I try this  there is an error "Error opening fd".

dagar, 2018-01-02:
If you open a pull request I can review and cross test. The I2C might be wrong for the raspberrypi as well. 

On Linux the i2c buses map to `/dev/i2c-X`.

belaidH, 2018-01-02:
okey, this is my pull requests  https://github.com/belaidH/Firmware/pulls

dagar, 2018-01-04:
Can you open as a pull request against the main PX4 Firmware repository? https://github.com/PX4/Firmware

belaidH, 2018-01-04:
yes, I did it.  https://github.com/PX4/Firmware/pull/8579

dagar, 2018-01-22:
Done!

closed 2018-01-22 by dagar