PX4/PX4-Autopilot #8540: Convergence crashes

LorenzMeier, 2017-12-30:
The first two flights of our Convergence after installation of Pixfalcon were failures. It flew ok in multirotor mode, but it didn't do well in airplane mode.

Control surfaces deflected correctly on the ground prior to flight in stabilize mode in multirotor and both stabilize and manual modes in airplane mode.

If anyone can do an autopsy of the logs, here they are.
Log of first flight:
https://logs.px4.io/plot_app?log=4acb5cae-f192-422d-8228-75ebec5f0c7b1
Following transition into airplane mode, throttle cut out, then returned to full throttle, then began to roll rapidly.

Log of second flight:
https://logs.px4.io/plot_app?log=d772eac7-7bde-4b42-bb95-5892078982241
Following transition into airplane mode, it pitched down and rolled right. Unable to counter with rc inputs. Crashed hard and GPS was found 30 feet from crash site.

@tumbili Do you have an idea what is going wrong here?

LorenzMeier, 2017-12-30:
@bkueng Flight review fails to display these logs - any insights?

RomanBapst, 2018-02-11:
I assume not many people have tried flying the convergence fully manually since we have a ready config for it.

RomanBapst, 2018-03-20:
We tested the convergence and it's fully functional. The position controller back-transition are very rough. This is something we will fix in future but we need to bring is some architecture changes for that.

closed 2018-03-20 by RomanBapst