PX4/PX4-Autopilot #8541: Mission: MAV_CMD_DO_SET_ROI not working

DonLakeFlyer, 2017-12-31:
* Use latest QGC daily. You need this fix: https://github.com/mavlink/qgroundcontrol/pull/5931. (may need to wait till tomorrow for it to make it all the way through the system). 
* Create a simple mission with a couple waypoints
* Turn one of them into an ROI (Under Camera category)
* Send to Vehicle

Two problems:
* If you fly the mission the vehicle will not point to the ROI when it is supposed to
* If you download the mission back from the vehicle. Param 1 of MAV_CMD_DO_SET_ROI command has been changed from 3:MAV_ROI_LOCATION to 0:MAV_ROI_NONE.

Tested using master Firmware and SITL+Gazebo.

dagar, 2017-12-31:
@ndepal FYI

ndepal, 2017-12-31:
I can't test this right now but I've had this working.

A couple of things that come to mind :
- Do you want the vehicle to yaw and not just the camera? If so, is the mission yaw mode set accordingly?
- heading towards ROI only works if it's given with the correct frame (global?) and qgc might not default to that one.

I'll be back in the office later this week, I'll test it out then.

LorenzMeier, 2018-01-01:
@ndepal I would appreciate if you could take the lead on this one. Thanks!

DonLakeFlyer, 2018-01-02:
I may have screwed this up now that I read your comments. Let me look again. I was running Lorenz's new gaazebo pull which has the gimbal working.

ndepal, 2018-01-03:
Tested on master (757d905089e277aa94c10c38b0b4c1bf48d97f17) with QGC Daily 93c9e91.

Works as expected.

The only thing is that QGC does not yet know about `MIS_YAWMODE` "heading towards ROI", which I had to enter manually and force save.

DonLakeFlyer, 2018-01-03:
If there is the requirement to change a parameter to get a mission to work then I would consider this not working. If you get an ROI command in a mission then this should override the parameter setting for that mission.

ndepal, 2018-01-04:
The reason the parameter is there now is that we not all users will want the vehicle to turn towards the ROI.  If you have a gimbal that can yaw you may want to the vehicle to face forward and only he camera to turn.

The `MIS_YAWMODE` "heading towards ROI" could be thought of a "Gimbal has no yawing capability" setting.

DonLakeFlyer, 2018-01-04:
> The MIS_YAWMODE "heading towards ROI" could be thought of a "Gimbal has no yawing capability" setting.

Doesn't the firmware know if the gimbal has yaw capability somewhere else based om the actual gimbal setup not this seemingly random parameter.

ndepal, 2018-01-04:
I don't know. If it does, I agree that that would be more elegant.

DonLakeFlyer, 2018-01-04:
Here is how I think it should work:
* Whether the gimbal supports roll/pitch/yaw should be unrelated to the MIS_YAWMODE parameter
* The MIS_YAWMODE parameter simply sets the default flight state for a mission
* If you get an ROI command the vehicle responds appropriately by either yawing the vehicle if the gimbal does not support yaw. Or adjusting the gimbal if the gimbal supports it.
* When a mission then specifies an ROI_NONE the vehicle returns to the default state as specified by MIS_YAWMODE
* If you want to be able to specify that the vehicle should never yaw for ROI even though the vehicle supports it then that requires a new additional parameter

This way things work out of the box without the user having to futz with things in order to get an ROI mission to work.
* If you want

DonLakeFlyer, 2018-01-04:
More:
* If the vehicle is tracking and ROI and the mission specifies another gimbal angle using DO_MOUNT_CONTROL then the vehicle falls out of ROI mode

ndepal, 2018-01-04:
> * Whether the gimbal supports roll/pitch/yaw should be unrelated to the MIS_YAWMODE parameter
> * The MIS_YAWMODE parameter simply sets the default flight state for a mission
> * If you get an ROI [...]

These changes should be relatively easy to incorporate. The main change would be to take the "Heading towards ROI" mode from `MIS_YAWMODE` and create a new boolean parameter for that, i.e. a "Gimbal has yawing capability" parameter. Probably in the `vmount` module?

ndepal, 2018-01-05:
While working on this, I've been struggling to get the [gimbal simulation](https://dev.px4.io/en/advanced/gimbal_control.html) to do what I expect.

When I upload a mission with an ROI, the gimbal turns as the vehicle flies, but it seems to yaw in some arbitrary direction, rather than towards the ROI. @DonLakeFlyer does this work well for you?

ndepal, 2018-01-05:
@DonLakeFlyer I've made a PR with the changes you proposed. Can you test it out?

I created a new parameter in `vmount` that specifies the gimbal's yaw capabilities.
I don't really like this much better. It's also not very intuitive because it only affects ROI commands in missions, and not other commands to `vmount` such as `MAV_CMD_DO_MOUNT_CONTROL`.

DonLakeFlyer, 2018-01-05:
FYI: I haven't gotten the gimbal to work in gazebo completely correct yet. Lorenz put in a fix a week back which makes it better but last I tested it still wasn't quite right. Haven't had time to look again. Let me try it again today. Then I'll ping your guys with my results.

closed 2018-05-27 by LorenzMeier