PX4/PX4-Autopilot #8556: Offboard control CI test unstable

LorenzMeier, 2018-01-02:
Fail example:
https://logs.px4.io/plot_app?log=3a1bf335-a6d7-41c8-a2b8-0b8dd4658ee0

Observations:

  * System never took off
  * Setpoints were there, but the actuators didn't respond
  * No actuator outputs plot present at all

Pass example (same commit):
https://logs.px4.io/plot_app?log=5331c5d9-ea98-489c-86ae-06a0344f92af

Script:
https://github.com/PX4/Firmware/blob/master/integrationtests/python_src/px4_it/mavros/mavros_offboard_attctl_test.py

dagar, 2018-01-02:
@lamping7 @AndreasAntener FYI

dagar, 2018-01-02:
Instructions to run locally:
``` bash
make posix_sitl_default && make posix_sitl_default sitl_gazebo
./test/rostest_px4_run.sh mavros_posix_tests_offboard_attctl.test
```

You need ROS and Gazebo installed, or you can do this within the px4io/px4-dev-ros docker container (https://hub.docker.com/r/px4io/px4-dev-ros/).

lamping7, 2018-01-02:
We talked about his already -- https://github.com/PX4/Firmware/pull/8447#issuecomment-354547441

Was anything done to try and fix it when I mentioned this last?

LorenzMeier, 2018-01-02:
No. What part of the home position changes do you think make it fail?

LorenzMeier, 2018-01-02:
Ok, I see. The home position is published even though its not the same level of "system readiness" as it was before.

LorenzMeier, 2018-01-02:
On reviewing the code I didn't find anything that explains the CI failure - I tried to Harden a few other things though:

https://github.com/PX4/Firmware/pull/8557

dagar, 2018-01-06:
Hitting various intermittent failures in the mission tests as well.
http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/PR-8605/1/pipeline

Overall failure rate is low, but we should definitely gets these stable.

@TSC21 any help would be greatly appreciated.

lamping7, 2018-01-07:
This should help figure out what is going on -- https://github.com/PX4/Firmware/pull/8616

dagar, 2018-01-07:
ROS vtol mission old 2 failure with logs.
http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/PR-8622/2/pipeline

dagar, 2018-01-09:
http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/PR-8639/2/pipeline

dagar, 2018-01-09:
http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/PR-8639/3/pipeline/726

dagar, 2018-01-16:
More testing. https://github.com/PX4/Firmware/pull/8699

lamping7, 2018-08-09:
This has been resolved.

closed 2018-08-09 by lamping7