PX4/PX4-Autopilot #8562: Build System does not build uORB headers for example apps

hamishwillee, 2018-01-02:
The helloworld example (and probably others) can be built "stand alone" or as part of a full build. If built stand-alone from a clean system the build fails because the uORB headers have not yet been generated. 

The build system should detect if headers are present, and if not either generate them or warn that they need to be generated. I assume this should be done as some sort of "global" check?

Some discussions:
- http://discuss.px4.io/t/no-such-file-or-directory-uorb-topics-sensor-combined-h/1336/8
- https://px4.slack.com/archives/C581ST17V/p1514839763000075

LorenzMeier, 2018-01-04:
@bkueng Could you please investigate?

bkueng, 2018-01-09:
@hamishwillee I don't quite understand what is meant with "stand alone". The [dev-guide](https://dev.px4.io/en/tutorials/tutorial_hello_sky.html) does not mention that, and I don't see how uORB should be of any use if you only have your own single app.

hamishwillee, 2018-01-09:
@bkueng I'm just the messenger here. I *think* the use case was that they built the app without having first build the whole of PX4. As you say, perhaps that is "not supported" so we can just close this.

Probably worth you looking at the discussion on slack and seeing if I have misinterpreted what is happening.

bkueng, 2018-01-10:
Hmm, reading slack I still don't understand the use case. If you want a reduced build, I suggest copying an existing cmake target and remove most of the modules. That will still give you a valid build as long as you don't remove uORB.

hamishwillee, 2018-01-10:
I'm going to close this then. The user has a link and can reopen with more detail if they can provide it.

closed 2018-01-10 by hamishwillee