PX4/PX4-Autopilot #8563: FW_ARSPD_MODE param meta data confusing

DonLakeFlyer, 2018-01-03:
So I went to fix this QGC bug: https://github.com/mavlink/qgroundcontrol/issues/5478. Here is what you see in the parameter editor:
[image]

It's confusing as to what Enabled/Disabled means. I would suggest switching the meta data to enum strings which are more descriptive. Something like:
* No airspeed sensor
* Airspeed sensor installed

Or something like that.

DonLakeFlyer, 2018-01-03:
Or in other words the concept of "Enabling" something in order to "Disable" it is odd.

dagar, 2018-01-03:
This happened when it went from a parameter with 3 options to a bool.

commit 4cb195a6 (2018-01-03): FW improve FW_ARSP_MODE metadata options

 - fixes #8563
PR #8564 (2018-01-03): FW improve FW_ARSP_MODE metadata options
 - fixes #8563
commit 757d9050 (2018-01-03): FW improve FW_ARSP_MODE metadata options

 - fixes #8563
closed 2018-01-03 by LorenzMeier