PX4/PX4-Autopilot #8567: EKF Replay generates 6 files instead of one

CarlOlsson, 2018-01-03:
I want to try out the uLog EKF replay but when I follow the [docs](https://dev.px4.io/en/debug/system_wide_replay.html#ekf2-replay) I get for me unexpected results.

The [logfile](https://logs.px4.io/plot_app?log=2f967654-4766-4732-be75-53e8e705e732) is from a quad running v1.7.2, u3 ultra SD card and SDLOG_MODE = 2 (logging from boot until shut down)

I run the replay using
```
make posix none
```

However, for some reason I get 6 replay files instead of one
```
INFO  [uorb] Using orb rules from ./rootfs/orb_publisher.rules
Sleeping for 0.2 s; (0 us).
INFO  [logger] logger started (mode=all)
INFO  [replay] using replay log file: /home/carl/Documents/logs_quad_home/2018-01-03/11_28_36.ulg
INFO  [logger] log root dir created: rootfs/fs/microsd/log
pxh> INFO  [replay] Ekf2 replay mode
INFO  [replay] Applying params from ULog file...
INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2018-01-03/19_26_39_replayed.ulg
INFO  [replay] Replay in progress...
INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
INFO  [lib__ecl] EKF commencing GPS fusion
INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2018-01-03/11_30_25_replayed.ulg
INFO  [logger] Start file log
INFO  [logger] closed logfile, bytes written: 3812493
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2018-01-03/11_30_44_replayed.ulg
INFO  [logger] Start file log
INFO  [logger] closed logfile, bytes written: 3481662
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2018-01-03/11_31_02_replayed.ulg
INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2018-01-03/11_31_41_replayed.ulg
INFO  [logger] Start file log
INFO  [logger] closed logfile, bytes written: 2353681
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2018-01-03/11_31_52_replayed.ulg
INFO  [replay] Replay done (published 53376 msgs, 2.832 s)
INFO  [replay] 
INFO  [replay] Topic, Num Published, Num Error (no timestamp match found):
INFO  [replay] sensor_combined: 52417, 0
INFO  [replay] vehicle_gps_position: 1059, 0
```

```
✔ ~/src/standard_PX4/Firmware/build/posix_sitl_default_replay/tmp/rootfs/fs/microsd/log/2018-01-03 [v1.7.2|✔] 
20:27 $ ls
11_30_25_replayed.ulg  11_31_02_replayed.ulg  11_31_52_replayed.ulg
11_30_44_replayed.ulg  11_31_41_replayed.ulg  19_26_39_replayed.ulg
```

Is this intended behavior?

bkueng, 2018-01-08:
It is expected but not necessarily intended. On replay, logger will create a new log on each (dis)arm. Can you test adding `-f` to https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris_replay#L4?

CarlOlsson, 2018-01-14:
@bkueng yes adding the -f results in one log file

closed 2018-01-15 by LorenzMeier