PX4/PX4-Autopilot #8568: Proposal - Parameter Meta Data: Categories

DonLakeFlyer, 2018-01-03:
Add support for each parameter to belong to a category as specified by the meta data. This will allow QGC to add an additiona level of hierarchy to parameter display hopefully making more discoverable to the user.

The proposed set of categories is: System, Developer, Advanced, Standard. From QGC side parameters which are mis-categorised or not have category specified will go into the Standard category.

The parameter meta data will change as follows:
    <parameter ... category="Standard">

dagar, 2018-01-03:
First and foremost I'd want to hide the parameters that a user would never manually set. Would this be System or something else? I'd argue some of these shouldn't even be sent to the ground (or stored as parameters to begin with), but that's another discussion.

**System (or another category)**
 - sensor calibration: most CAL_*
 - temperature compensation: TC_*
 - vehicle flight time
 - UUID
 - EKF learned bias
 - RC calibration? Slightly different case, as these are set by the QGC RC calibration routine.

**Advanced**
 - circuit breakers
 - per sensor enable/disable

FYI @sanderux

sanderux, 2018-01-03:
How about
- Basic
- Expert
- System

dagar, 2018-01-03:
Possibly more of a QGC discussion, but perhaps any param that's already exposed on a friendly setup tab shouldn't be shown in the tree (or maybe those are what we consider Standard).

Between that and hiding all the advanced/system params, the remaining set of user intended parameters that aren't already accessible might be pretty manageable?

hamishwillee, 2018-01-03:
> First and foremost I'd want to hide the parameters that a user would never manually set.

Yes! They should be invisible by default (require some sort of explicit action that always default off to view). 

> but perhaps any param that's already exposed on a friendly setup tab shouldn't be shown in the tree (or maybe those are what we consider Standard)." 

I don't think so. Anyone who wants to work with parameters expects them all to be there - frustrating if they aren't and then you have to work out what screen they are really on. I'd be happy for there to be a filter to hide these though. 

I don't have a strong feeling about what the categories are as long as we can get the filtering above - ie System has to mean "should not be set by a user directly". @sanderux set seems sensible

DonLakeFlyer, 2018-01-03:
I'll let you guys duke it out over naming. I don't think there should be a hidden from the user set. That just turns in a pain in the ass. Because then you need a way to unhide it and so forth. All categories and parameters should be visible.

hamishwillee, 2018-01-03:
If not invisible, then perhaps colour coded (greyed) to indicate that they are not really intended for use?

dagar, 2018-01-04:
Alright, then at the risk of derailing the metadata conversation I think it's time to talk about acceptable param system usage. Many of these "system" params that I wanted to hide should not be user editable in the first place. Additionally the system auto updating some of them every flight breaks things like the param HASH_CHECK which can skip the full param sync on first connect.

dagar, 2018-01-04:
Other than the generated parameters (https://github.com/PX4/Firmware/issues/8570) and a handful of obvious examples (circuit breakers) the last time we discussed this I actually found it hard to think of a good line between basic and expert. What's everyone else thinking?

One extreme to think about is if a user is handed a completely configured system. What parameters do they need to be able to edit?

Another is a fresh setup on a custom vehicle. Is the bare minimum they *need* to configure in the basic set?

LorenzMeier, 2018-01-04:
WIP for that here: https://github.com/PX4/Firmware/pull/8577

closed 2018-08-10 by santiago3dr