PX4/PX4-Autopilot #8574: Understanding ekf2 code

ajaybhatt003, 2018-01-04:
Hello,
I am a graduate student. I have just started working on ekf2 (https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_main.cpp) code of px4 firmware. Kindly can you help me to understand it.
I am currently referring to this site http://ardupilot.org/dev/docs/ekf2-estimation-system.html but not sure that the same code is written in cpp as ekf2 for px4 firmware.
Please reply when you get time if possible provide me literature which explains about ekf2 code so that I can convert it to matlab code for my purpose.
Thanks in advance.

LorenzMeier, 2018-01-04:
You will want to have a look at Matlab here: https://github.com/PX4/ecl/tree/master/matlab

closed 2018-01-04 by LorenzMeier

hamishwillee, 2018-01-04:
IN addition, EKF Documentation is in the PX4 Development Guide [here](https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html).

ajaybhatt003, 2018-01-04:
Thank you for prompt reply to both of you. 
@hamishwillee please can you explain EKF documentation you mean  http://ardupilot.org/dev/docs/ekf2-estimation-system.html .

@LorenzMeier can you please point out to me any particular code to see or can you pass any documentation that can help me to understand the structure of code of ekf2 in px4 firmware.

Thanks

LorenzMeier, 2018-01-04:
No, the documentation is here: https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html

ajaybhatt003, 2018-01-04:
Thank you @LorenzMeier for replying.
Kindly can you provide me information that the ekf_replay codes within https://github.com/PX4/ecl/tree/master/matlab after addition makes the entire ekf2 code?
Thanks

LorenzMeier, 2018-01-04:
This is an issue tracker. Please feel free to ask questions and discussions on http://discuss.px4.io. Thanks!