PX4/PX4-Autopilot #8578: Support battery status without adc

ChristophTobler, 2018-01-04:
Currently there is no obvious way of supporting a power module that does not use adc https://github.com/ChristophTobler/Firmware/blob/master/src/modules/sensors/sensors.cpp#L414 (see e.g. https://github.com/PX4/Firmware/pull/8576).

What would be a good way of supporting both and not too much code copying?

@dagar ideas?

ChristophTobler, 2018-01-04:
Ok. Then I guess I can leave https://github.com/PX4/Firmware/pull/8576 as is (short term).

ChristophTobler, 2018-01-17:
FYI @dagar @MaEtUgR @davids5

dagar, 2018-01-17:
**Pieces to consider**
 - nuttx ADC
  - N power modules (analog voltage + current)
  - system_power

 - aero i2c adc?
 - crazyflie
 - i2c smart battery (smbus)
 - snapdragon smart battery

Which of these cases need calibration parameters?
Who decides the overall system battery percentage estimate?