PX4/PX4-Autopilot #8585: Intel Aero not retaining parameters after reboot

santiago3dr, 2018-01-04:
When flashing master 430cdad calibrations on all sensors and radio are required (1).
Successfully ran Gyro calibration (2).
Once calibrating and rebooting, all calibration values are lost (3).
Reverting to yesterdays master 757d905 all calibration values are restored WITHOUT having to re-calibrate (4).
[image]
[image]
[image]
[image]

dagar, 2018-01-04:
Could you go to the param tree and do a full refresh? Now try to change a random parameter and save. After a reboot is that parameter preserved?

santiago3dr, 2018-01-04:
Flashed 430cdad and went to parameter tree, hit refresh, changed NAV_MC_ALT_RAD from 0.8 to 1.1
Did a power cycle on the vehicle and parameter reverted to default 0.8

dagar, 2018-01-04:
Thanks Santiago. So it seems I broke flash based parameter writing (intel aero only I believe), but it's still able to read successfully. I suspect this hack is the culprit - https://github.com/PX4/Firmware/commit/430cdada602ee5037f7b8f5c43b615e181400351

dagar, 2018-01-04:
https://logs.px4.io/plot_app?log=5a8b3640-a795-4386-ae2f-ec5c43fa2264

[image]

LorenzMeier, 2018-01-04:
Looks like the wrong file descriptor since Intel Aero does not have an SD. Board detection issue?

dagar, 2018-01-04:
I think a lot of the error messages aren't quite right for the Aero. This should be fixed in https://github.com/PX4/Firmware/pull/8586

dagar, 2018-01-05:
Surprisingly this doesn't seem to be related to the tinybson change.

1. Test build 1 (commit right before the bson encoder buffer)
   - https://github.com/PX4/Firmware/commit/08443c0
   - https://github.com/PX4/Firmware/compare/08443c0...master
2. Test build 2 (commit right before all recent param changes)
   - https://github.com/PX4/Firmware/commit/c8590e0
   - https://github.com/PX4/Firmware/compare/c8590e0...master

Which means the problem is somewhere in here. https://github.com/PX4/Firmware/compare/c8590e0...08443c0

dagar, 2018-01-05:
The perf counters seem to be the problem on aero.

dagar, 2018-01-12:
The header ordering somehow broke params on the Aero.

commit c2ce8692 (2018-01-05): params tinybson restore buffer length

 - fixes #8585
commit 58492856 (2018-01-12): param restore header ordering

 - fixes #8585
commit 1dddecfc (2018-01-12): param include px4_config.h for FLASH_BASED_PARAMS

 - fixes #8585
commit a25ca0c3 (2018-01-12): param include px4_config.h for FLASH_BASED_PARAMS

 - fixes #8585
closed 2018-01-12 by LorenzMeier