PX4/PX4-Autopilot #8590: Gazebo build broken

ndepal, 2018-01-05:
`make posix_sitl_default gazebo_iris`  is broken on master:

```
/home/nicolas/src/testing/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp: In member function ‘void gazebo::GeotaggedImagesPlugin::handle_message(mavlink_message_t*)’:
/home/nicolas/src/testing/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp:353:50: error: too many arguments to function ‘uint16_t mavlink_msg_command_ack_pack_chan(uint8_t, uint8_t, uint8_t, mavlink_message_t*, uint16_t, uint8_t, uint8_t)’
                                        msg.compid);
                                                  ^
In file included from /home/nicolas/src/deps_catkin_ws/devel/include/mavlink/v2.0/common/common.h:1165:0,
                 from /home/nicolas/src/deps_catkin_ws/devel/include/mavlink/v2.0/common/mavlink.h:32,
                 from /home/nicolas/src/testing/Firmware/Tools/sitl_gazebo/include/gazebo_geotagged_images_plugin.h:21,
                 from /home/nicolas/src/testing/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp:18:
/home/nicolas/src/deps_catkin_ws/devel/include/mavlink/v2.0/common/./mavlink_msg_command_ack.h:89:24: note: declared here
 static inline uint16_t mavlink_msg_command_ack_pack_chan(uint8_t system_id, uint8_t component_id, uint8_t chan,
                        ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:147: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1
```

A bisect indicates https://github.com/PX4/Firmware/commit/e3f5f8e475b550f1ac1779c1d87d33030fc86609 breaks it.

TSC21, 2018-01-05:
When you build `sitl_gazebo` have you checked where is it getting the Mavlink headers? Currently the FindMavlink.cmake looks for the headers in the system files (it's how `find_path` works but we can tweak other way). So if your `ros-<$ROS_DISTRO>-mavlink` is not updated, you may get the above error. I don't get the above issue building master on my machine and CI is currently passing on master.

TSC21, 2018-01-05:
I will update `find_path()` on the `FindMavlink.cmake` so he finds the packages in a different order (it doesn't make a difference currently the order they are disposed).

ndepal, 2018-01-05:
Thanks for having a look. Indeed it seems like it finds a mavlink version in one of my ROS workspaces.

I definitely think it should use the mavlink version already available in Firmware.

TSC21, 2018-01-05:
The way it finds is more generic now than it was before, allowing for using the package without depending on PX4. Just have to change what it searches and fins first.

ChristophTobler, 2018-01-05:
Same here :+1: I also think it would make sense to look for the version available in the Firmware first.

TSC21, 2018-01-05:
That will be done.

dagar, 2018-01-05:
Resolved in https://github.com/PX4/Firmware/pull/8597.

closed 2018-01-05 by dagar