PX4/PX4-Autopilot #8592: Unfiltered IMU data for apps outside Firmware

ChristophTobler, 2018-01-05:
There's currently no proper solution to get unfiltered IMU data outside the Firmware.
https://github.com/PX4/Firmware/pull/8591 is an example hack for the Snapdragon Flight.

Ideas:
- Create new message for unfiltered data (using floats)
- Drivers publish unfiltered data and the consumer have to filter according their needs

MaEtUgR, 2018-01-08:
The three cases for which I see raw a IMU raw data output requirement reoccuring:
- debugging (where maybe logging would be enough)
- research projects that deal/prototype with IMU data but want to run "offboard"
- existing algorithms which are designed to directly consume the raw data and cannot work with processed state estimates

VISLAM looks like it is the last case. So the feature even though it's probably not for a huge audience probably makes sense to simplify developement.

dagar, 2018-10-03:
@ChristophTobler could you expand on your requirements?

Are delta angles and delta velocities desired?
At what rate?

mhkabir, 2018-10-03:
This is a duplicate of https://github.com/PX4/Firmware/issues/9796. Closing in favour of that.

closed 2018-10-03 by mhkabir