PX4/PX4-Autopilot #8599: Gazebo Gimbal: Working incorrectly

DonLakeFlyer, 2018-01-05:
* Load the attached plan in Gazebo SITL gazebo_typhoon_h480
* The Plan file should alternate gimbal yaw from 90 to -90 at each waypoint 
* Start Mission
* You'll see that as the vehicle goes up the gimbal is pointing forward which is correct
* But when the vehicle turns towards the takeoff waypoint the gimbal is now at some random angle
* As it transitions through the first waypoint the gimbal does not yaw when it should
* As it transitions through the remaining waypoints the gimbal does yaw, but the direction is reversed from where it should be

[GazeboGimbal.plan.zip](https://github.com/PX4/Firmware/files/1607598/GazeboGimbal.plan.zip)

DonLakeFlyer, 2018-01-05:
@LorenzMeier @ndepal Gazebo gimbal testing results

ndepal, 2018-01-08:
Tested this on https://github.com/PX4/Firmware/pull/8620.

> But when the vehicle turns towards the takeoff waypoint the gimbal is now at some random angle

I don't see hat issue. The gimbal points forward until WP 2 is reached.

> As it transitions through the first waypoint the gimbal does not yaw when it should

This is fixed by the PR:
https://github.com/PX4/Firmware/pull/8620/files#diff-357c573ea1368bd031da4870d1fbb1feR380

> As it transitions through the remaining waypoints the gimbal does yaw, but the direction is reversed from where it should be

This appears to be a coordinate frame issue. PX4 is in NED and the roll/pitch/yaw values are simply copied from the mavlink msision item.
Can you confirm that:
- positive yaw angles entered by the user should turn the camera counter clock-wise when looking from the top down?
- positive pitch angles should turn the camera up?
- positive roll angles should turn the camera clockwise when looking through the camera?

DonLakeFlyer, 2018-01-12:
Confused by what look like real firmware changes to get the gimbal to point correctly in gazebo. I had thought the gimbal was working fine on a real vehicle. Only Gazebo was wrong. Maybe I'm wrong there.

DonLakeFlyer, 2018-01-12:
I tried testing with pull #8620 and things got even stranger:
* Still got random gimbal angle while flying to takeoff waypoint
* The gimbal transitioned at each waypoint but they were reversed
* WHen I downloaded the mission back from the vehicle I found that the gimbal angles I had sent had been reversed. I sent 90, -90, 90. When I downloaded they had been changed to -90, 90, -90.

ndepal, 2018-01-12:
> The gimbal transitioned at each waypoint but they were reversed

This is what you observed before #8620, so it's still the same?

>WHen I downloaded the mission back from the vehicle I found that the gimbal angles I had sent had been reversed. I sent 90, -90, 90. When I downloaded they had been changed to -90, 90, -90.

I'll have to look at what happens when you download the mission from the vehicle. I might have to revert the coordinate transformation there.

DonLakeFlyer, 2018-01-12:
Correct on gazebo gimbal visuals reversed being the same as before.

ndepal, 2018-01-15:
> Correct on gazebo gimbal visuals reversed being the same as before.

Did you re-upload the mission after checking out this PR?
The sign change is applied when processing an mission that's being uploaded.

Here is a video of what I see. It is the same mission that you uploaded above, just with the waypoints moved closer.

https://drive.google.com/file/d/1KNQUeVi6Yrx7O3pFkvd0OVmhzXdHbv_K/view?usp=sharing

The gimbal yawing seems to be correct in the sense that the gimbal faces in the same direction as the white indicators on the waypoints (assuming those should indicate gimbal direction?).

What's confusing me right now is that the white indicators switch sides when switching from the mission editor to the fly tab.

Can we take a step back and answer the above question?
>positive yaw angles entered by the user should turn the camera counter clock-wise when looking from the top down?

DonLakeFlyer, 2018-01-15:
Let me try again. Something seems out of whack on QGC side with the plan/fly swapping directions.

DonLakeFlyer, 2018-01-15:
Ok, so aside from the plan/fly view out of whack. I believe I may have screwed up on the yaw angles. I forgot how we had worked out something for pitch angles between QGC and mission item. 

They work like this from a QGC/user perspective:
* User can enter values from 0 to 90 degrees
* 0 indicates level
* 90 indicates pointing straight down at the ground

But here is what QGC understands/sends in the mission item:
* QGC modifies these values when storing to/loading from mission item by multiplying by -1
* So vehicle sees values in range of -90 to 0 degrees
* On vehicle side 0 means level
* On vehicle side -90 means pointing straight down

So starting from scratch on yaw angles from QGC / (from a user perpsective):
* User can enter values from -180 to 180
* 0 indicates straight forward
* 90 indicates clockwise rotation
* -90 indicators counter-clockwise rotation

Which leaves the question as to what should be in the mission item? I can use the same logic to reverse (multiple by -1) the yaw values as well just like pitch.

DonLakeFlyer, 2018-01-15:
Also would be good to update the mavlink spec with all the details we come up with here. Current mavlink spec on DO_MOUNT_CONTROL says nothing about this stuff.

DonLakeFlyer, 2018-01-15:
Fix for gimbal angle QGC visuals is here: https://github.com/mavlink/qgroundcontrol/pull/6024. Just a visual display problem.

ndepal, 2018-01-15:
I would vote for sending yaw values where positive values mean a clockwise rotation (i.e. no sign change).

This would be consistent with the NED frame and would make the sign change I added unnecessary: https://github.com/PX4/Firmware/pull/8620/files#diff-0eafe063c4fc776dc3890496a001ddc7R1437

DonLakeFlyer, 2018-01-15:
QGC is sending, positve angles indicates clockwise now. So that's current reason the gimbal is pointing the wrong direction. If you take that out, then I think everything will start working correctly.

ndepal, 2018-01-16:
@DonLakeFlyer I have removed the sign change in https://github.com/PX4/Firmware/pull/8620. Please test again and see if it now works as expected.

DonLakeFlyer, 2018-01-16:
My test results:
* When I fly the mission the gimbal reacts correctly to the gimbal commands.
* But the initial gimbal position after takeoff is still wacky (see video: https://drive.google.com/open?id=1wtAh7B8Q8PmegMuPtzRgfl2KRDVeUIQa). Slightly off to the left. It corrects when it gets the first gimbal command.
* When I download the misison back from the vehicle it is correct.

ndepal, 2018-01-16:
>But the initial gimbal position after takeoff is still wacky

I believe this wackiness is a problem specific to the Gazebo gimbal plugin and not a fimware issue. You can check that `vmount` is outputting 0deg commands as it should with a print statement here: https://github.com/PX4/Firmware/blob/4453e4201b7a245cff52beeb38a293161aea4c48/src/drivers/vmount/output_rc.cpp#L78

Does #8620 now resolve https://github.com/PX4/Firmware/issues/8541?

DonLakeFlyer, 2018-01-19:
> Does #8620 now resolve #8541?

I believe so. It's hard to test again now since QGC has cut over to new ROI commands and this pull is on old commands. I think the thing to do is merge your pull in. Then I can rebase my new ROI command to your changes and fix up as needed. Then I can retest the whole thing end to end.

Question for testing: If I want to test using SITL a vehicle with and without gimbal yaw capability how do I do that?

ndepal, 2018-01-19:
Ok, I will remove commit https://github.com/PX4/Firmware/pull/8620/commits/bb50a3ec56742647674bee7532e6c4924784f447 from #8620 in favor of https://github.com/PX4/Firmware/pull/8734 and rebase on monday.

> Question for testing: If I want to test using SITL a vehicle with and without gimbal yaw capability how do I do that?

You can test both with the typhoon by setting `MIS_MNT_YAW_CTL` appropriately. To test `MIS_MNT_YAW_CTL` disabled any vehicle will work.

DonLakeFlyer, 2018-01-20:
Thanks @ndepal. If this gets merged in on Monday I should be able to test Monday as weell. Otherwise starting Tuesday I will be out of town for a week. So bug delay.

LorenzMeier, 2018-01-21:
I just merged https://github.com/PX4/Firmware/pull/8734 - @ndepal Would be awesome if you could rush this through Monday so we can wrap it up.

DonLakeFlyer, 2018-01-30:
I just tested this again against master. Everything works now except for the gimbal doesn't stay pointed straight forward in between takeoff and waypoint 2. It yaws off to some small angle.

ndepal, 2018-01-31:
I'm pretty sure what you're describing is a peculiarity of the simulated gimbal. I see it oscillating quite a bit as well. This has nothing to do with the PX4 code though.

DonLakeFlyer, 2018-02-02:
I believe this is all good now

closed 2018-02-02 by DonLakeFlyer

tawfiq1200, 2021-08-12:
Cannot move the gimbal using QGC or Command Line Interface following [this](https://docs.px4.io/master/en/advanced/gimbal_control.html) , seems like Gazebo plugin is not working, using Gazebo 11, any suggestions?