PX4/PX4-Autopilot #8600: Sudden throttle drops in altitude mode (release 1.7.0, 1.7.2)

Changliu52, 2018-01-05:
Hi all,

I have recently setup a generic X airframe, with all default settings.
In altitude mode, without stick input, I experienced occasional sudden throttle drops.

The setup envirenments:
Pixhawk2 
firmware 1.7.2 or 1.7.0
generic X airframe
default parameters.

As the plot shown below, it seems sudden vertical acceleration was commanded by the velocity controller for no reason, since the local velocity being pretty smooth:
[image]

All the sensors were functioning correctly without sudden jumps, which is shown in the full log here: https://www.dropbox.com/s/66d13znykptmvdc/log_16_UnknownDate.ulg?dl=0

Is this related to https://github.com/PX4/Firmware/issues/7443 as the velocity feedforward term?

Thanks in advance.

Chang

dagar, 2018-01-05:
Log uploaded to flight review. https://logs.px4.io/plot_app?log=8278c297-5342-496a-a6c8-3536308e6a49

dagar, 2018-01-06:
It's coming from the multicopter position controller (not manual input), I'm just not sure where yet. 

[image]

https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2858-L2866

dagar, 2018-01-06:
The land detector strikes again.

[image]

LorenzMeier, 2018-01-06:
Let's see if this is a setup issue. If it is, we might need a setup screen or some boundary checks. The land detector is a simplistic vehicle model and not setting that up correctly is asking for trouble.

Murcamil, 2018-01-07:
The problem that I and Changliu52 had has been solved, apparently (land detector not triggered for the past 4 full flights). Changes are:
1. Added the drones payload (about 1kg) - drone weights around 4kg 
2. Updated Hover throttle and Manual minimum throttle for new weight as showed below (increased both)
3. Increased P on all axis 
Parameters that are not default but I think are unrelated to the issue: MC_ROLLRATE_D, MC_ROLLRATE_MAX, MC_ROLLRATE_P, MC_ROLL_P, MC_PITCHRATE_D, MC_PITCHRATE_MAX, MC_PITCHRATE_P, MC_PITCH_P, MC_YAW_MAX, MC_YAW_P. MPC_LAND_SPEED = 0.3 m/s.

[image]
[image]

LorenzMeier, 2018-01-07:
Thanks, that's helpful feedback! The land speed of 0.3 m/s is problematic in itself.

Murcamil, 2018-01-07:
I am a first time user of the PX4, switching from 5 years of ArduCopter where this parameter was not problematic and only limited at 0.2 m/s if I remember correctly. I was prompted that 0.3 m/s was below the minimum but continued in thinking that I know what I am doing :smiley:

LorenzMeier, 2018-01-07:
The problem is that you left LNDMC_Z_VEL_MAX at its default of 0.5 m/s, which meant that your 0.3 m/s land speed were detected as "on the ground". I think I will add a pre-flight check to make sure these can't be set incorrectly.

LorenzMeier, 2018-01-07:
Here is the detection PR which will other users having the same issue warn about the parameter configuration issue: https://github.com/PX4/Firmware/pull/8618

closed 2018-01-07 by LorenzMeier