PX4/PX4-Autopilot #8601: Main PWM Glitch

tops4u, 2018-01-05:
**Bug Report**

When performing a Reset Commanded vom QGC or also on other occasions it may happen, that one Motor may start spinning for a split seconds and immediately stop - causing the Propeller to do about ¼ turn maximum. I have observed this a several occasions also today when I commanded a Mag Calibration via QGC (Radio Link). This also happened when IO Safety was enabled and Safety Switch has not yet been pressed. This will most of times affect a different motor (PWM Out). I assume there is a short signal Glitch on the line which is interpreted by the ESC as PWM Signal. ESC are HobbyWing Mini 35A BLHeli (Stock Firmware)

dagar, 2018-01-05:
Exactly which hardware? 

@davids5 FYI

tops4u, 2018-01-05:
Sorry - this is mRo X-2.1 with latest stable (1.7.2).

tops4u, 2018-01-06:
Checked this today and by chance got it on the right channel just after rebooting the Board. (Sorry for the Image Quality, this older Scope stores on vintage Floppy Disks only). As you can see the PWM commands 1600us just after booting. The Propeller did about 1 turn. 

[image]

tops4u, 2018-01-06:
After many, many, many Reboots an trying to get the Thingy on the Scope I succeded. Noticed by the way that my IO Processor crashed after some 30 Reboots and did not restart - also after a Reboot. Only Powercycling helped. 

Here is a Foto taken from the Scope (sorry for taking a Foto, this is a vintage LeCroy, the last device that has a Floppy Drive in our House). 

[image]

This shows a short PWM Signal of 1338us just after the reboot.

davids5, 2018-01-09:
@tops4u - I am not sure what I am looking at here. I would have expected one pulse. The issue in the past on the VMUv4 was fixed by changing the timing as done in this pr https://github.com/PX4/Firmware/commit/e86d9adb93189a3811a5ee38a2e4fadb7456423f adding board_on_reset as it is [here:](https://github.com/PX4/Firmware/blob/master/src/drivers/boards/px4fmu-v4/px4fmu_init.c#L166) I am not able to do any development at this time but a PR with the same logic would be welcome.

tops4u, 2018-01-09:
@davids5 This is the PWM Output of MAIN 1 just after issuing a Reboot via QGC. The PWM Signal will stay persistent throughout the Reboot. I think just after all is done FMU issues a Command to the PWM Outputs that may (or may not) lead to Pin being high for more than 1000us. The Problem will never occur during PowerUp, only when Rebooting the board. I have to recheck but AFAIR there was a short turn on a Motor also after Mag Calibration. 
Your solution may help for the Reboot Issue I think. This would put the PWMs low for the duration of Bootloader & Boot? If so, was it not the intention that if the Board would reboot in MidAir that it would resume Operation ASAP? So pulling the PWMs low would be a bad idea?

davids5, 2018-01-10:
Oh. The previous issue was with the AUX pins. I missed that this is on the Mains. This sounds like a state transition from running to running after running. I wonder if this is related to a recent change in the mixer logic on the PX4IO. I will have to get this on my bench when I can.

tops4u, 2018-01-12:
@davids5 anything I can do to help?

davids5, 2018-01-13:
@tops4u - Thank you for looking at this. There was a change on how the mixer is updated I think @LorenzMeier made to prevent allocation of memory under interrupt context off of the DMA interrupt. This may have created the issue. A test would be to revert that commit and see if the problem still persists.

tops4u, 2018-05-03:
@davids5 anything I could test here?

davids5, 2018-08-20:
@santiago3dr was this tested and does not have the issue any longer or just closed?

santiago3dr, 2018-08-20:
I see this issue was addressed in PR #10186 which has been merged
don't have the hardware to test this, should we leave it open?

davids5, 2018-08-20:
@tops4u - any possibility of a re-test?

tops4u, 2018-09-03:
@davids5  & @santiago3dr sorry to tell you that the retest failed. I still have intermittent PWM Spikes on the Main PWM. Approx every 3rd reboot. I think it also affects AUX Outputs. I have an custom ATMEGA328 Board that handles Lights & Gear which also reacts to reboots.

closed 2019-02-11 by stale[bot]