PX4/PX4-Autopilot #8602: RC Bus Issue when running on USB Power

tops4u, 2018-01-05:
**Bug Report**

When running only on USB Power a Disconnected RC Receiver or lost TX Signal causes the RC Lost Logic to go mad and also various Values to be interpreted into a non existing Signal.

[image]

Current Configuration : mRo-X2.1 with Futaba T10J and Futaba R3008SB (crosschecked with a R2008SB Receiver). This System is running the newer T-FHSS / S-FHSS System from Futaba. 

Earlier Configuration showing the same Problem was with AUAV-X2 and Spektrum DX7s / AR10000. 

As I have also had a crash due to a Kill Switch Command (which was most probably not sent from the TX) I have run a testflight today and have had heavy Problems controlling the UAV. I am aware that there are high Vibrations and I have to improve Mounting of the FC as well as Propeller Balancing. Still suddenly RC Signals went havoc and started switching Flight Modes as well as issuing Direction Commands - mostly due to a sudden YAW Movement I noticed a Problem. Luckly I could land the Copter in a nearby field - no Damage to the Copter, but the street lantern was a near miss. Even in Landed State the Kill Switch state was toggling. 

I have been logging RC Values with QGC for some time on the ground (unarmed) but was not able to reproduce the Problem. 

Flighltog (with awful Vibrations) can be found here: https://review.px4.io/plot_app?log=579d1e27-ada0-4fca-9c04-37aafa834312

LorenzMeier, 2018-01-06:
We do not have similar reports with other boards. @pkocmoud Could you send me the schematic via email and also run some tests on your end to see if you can reproduce this?

tops4u, 2018-01-06:
Schematics are here https://github.com/mRoboticsIO/Hardware/blob/master/X2.1/Docs/x21_V2_schematic.pdf

SIPI Signal passes a 74LVC2G240 IC before going to the FMU.  

If I read the Datasheet correctly the Schmitt Trigger will go High @2.0V and Low @0.8V when powered from 2.7-3.6V. So even when the RC Receiver will get only around 4.4V (indicated by Telemetry) it should still work. I can also perform some measurements or try with a replacement Unit, especially with the receiver, as this Unit was the one that fell down with a sudden Kill-Switch. I have also an Oscilloscope to check the signal.

tops4u, 2018-01-06:
Checked with another mRo-X2.1 Unit, where this problem did not happen. This may be a HW Issue, I'll check with the Supplier.

closed 2018-01-06 by tops4u

tops4u, 2018-01-08:
Figured this (the Flight Problem) to be most probably a stupid user fault, as Pinout for the Power Connector has changed from AUAV-X2 to mRo-X2.1 and I did not pay close attention. There was a Pin Swap for GND and CurSens - wondering why this even worked at all!

This does still not explain the USB Issue.