PX4/PX4-Autopilot #8603: Eagle: Auto and RTL non responsive

santiago3dr, 2018-01-05:
Running master c0c0666
Auto and RTL do not respond despite having good gps lock
Flight in position mode without issues
https://logs.px4.io/plot_app?log=474076d1-969e-41fd-bcf7-eaa0489c3038
https://logs.px4.io/plot_app?log=f9593388-cbae-48b9-a55d-4884dd162644

Logs not showing error messages

dagar, 2018-01-06:
Can you expand on what you mean by auto and RTL not responding?

These baro spikes are also a bit concerning, but not your main issue.
[image]

santiago3dr, 2018-08-01:
this issue has been resolved and both platforms work on 1.8.0 stable

closed 2018-08-01 by santiago3dr