PX4/PX4-Autopilot #8608: cmake for px4fmu-v4_default does not compile new files in init.d and mixer

lowjunen, 2018-01-06:
I noticed that whenever I compile and upload via <make px4fmu-v4_default upload>, the build file does not get updated with whatever new init.d and mixer files I added. I found a temporary fix by updating the px4_nuttx_configure command under cmake/configs/nuttx_px4fmu-v4_default.cmake to 

`px4_nuttx_configure(HWCLASS m4 CONFIG nsh ROMFS y ROMFSROOT px4fmu_common IO)`

Is this a proper fix or should I find another way around this problem?

Many thanks!

LorenzMeier, 2018-01-06:
Ok, I can reproduce this. File changes do re-generate the ROMFS, file additions don't.

LorenzMeier, 2018-01-06:
And after analysing this we could only "fix" it by requiring a re-generation of the airframes on every build, which would slow things down. I've instead opted to document this, as authors adding airframes need anyway need to know what they are doing:

https://dev.px4.io/en/airframes/adding_a_new_frame.html#configuration-file-overview

Thanks for reporting this, much appreciated! I'm sure you've saved a lot of other people time scratching their heads and you've helped us improving our user experience.

closed 2018-01-06 by LorenzMeier