PX4/PX4-Autopilot #8613: MAVLINK Flightmode not published after Reboot

tops4u, 2018-01-06:
**Bug Report** 

After a Reboot the Flightmode is not correctly transfered to QGC. It will show Manual. 

Just after Reboot (while RC FM-Switch being in Position Stabilized):

[image]

After switching from Stab -> PosCtl -> Stab:

[image]

dagar, 2018-01-06:
Do you have a corresponding log to verify? You could also take a look on the command line. `commander status`

tops4u, 2018-01-06:
TLog? Or what log file would you require?

tops4u, 2018-01-06:
Duplicate to #8268

tops4u, 2018-01-07:
Checked with listener commander_state and listener vehicle_status just after Boot Main State is 0 Nav State 0. After toggling the FM Switch on the TX Main State goes to 8, Nav State to 15.

dagar, 2018-01-07:
Reading the original description here again I'm now guessing the original switch to position was ignored (likely no valid position available).

I'm wondering if there's something we could do to better handle this case. Perhaps the RC state at bootup should be respected as long as a mode change hasn't been commanded (RC or Mavlink), or any attempted vehicle arming. Either the disarmed system could initialize the main_state based on bootup RC position, or we evaluate RC switch positions under certain conditions (rather than only on changes).

We have to be careful with automating changing modes to not do something even less intuitive.

tops4u, 2018-01-08:
Checked on Vehicle after Arming (Spinning Rotors) the Flightmode is not updated. How is this supposed to work? Respectively I would assume that then I Power RC and Vehicle that the FlightController would take whatever Flightmode is selected on the RC System.

tops4u, 2018-09-05:
AFAIK this is still the case, also with current Master.

closed 2019-02-19 by stale[bot]