PX4/PX4-Autopilot #8619: CI Mission Test Ideas

dagar, 2018-01-07:
First need a mechanism to describe full test
 - vehicle
 - parameters
 - starting location (lat, lon, alt)
 - mission file
 - extras like weather?
 - events that trigger failsafes (see below)

TODO
 - MC full mission coverage
 - FW full mission coverage
   - takeoff, land, loiter needed
 - VTOL all types
   - standard, tailsitter, tiltrotor (no model yet)
 - Test failsafes
   - low battery
   - loss of GPS
   - sensor failure (accel, gyro, airspeed)
   - engine failure (regular FW and VTOL)

hamishwillee, 2018-01-07:
I am interested in this, because I've had trouble getting clear information on how to set up various test conditions. You do the work in docker, and docs will benefit :-)

For example, sensor failures - add compass to that list. I think adding response to wind makes a lot of sense though possible as a separate set of tests since I'm not sure how useful it would be unless somewhat randomised.

How do you plan on mapping failsafes to position in a mission? For example, the new RTL landing behaviour on FW presumably needs to be tested in a number of ways at different positions in a mission cycle?

hamishwillee, 2019-01-28:
@dagar Should this reopen or did we get our testing story in a pile?

hamishwillee, 2019-10-08:
@dagar do you want to pin this?

Antiheavy, 2020-01-06:
@dagar Is this still relevant?  Is it still possible to help define fixed wing mission profiles?

closed 2025-02-14 by mrpollo