PX4/PX4-Autopilot #8624: Support for ICM-20948 as external I2C Compass

caomuqing, 2018-01-08:
Feature Request

The HERE GNSS module from Hex Technology comes with ICM20948 as an external compass, replacing the already EOL HMC5983 compass. Can we request adding support for that?
Thanks!

LorenzMeier, 2018-01-08:
Sure! Not hard, but we don't have hardware with any of our core developers right now.

caomuqing, 2018-01-08:
Thanks Lorenz, I can ask to send samples to developers. May I ask who we should send to, and the shipping address?

LorenzMeier, 2018-01-08:
Can you send me an email to lorenz@px4.io? Thanks!

mrlarsendk, 2018-02-14:
Hi @LorenzMeier,

Are there any status updates on this issue? Do you have the hardware at hand?

Best regards,
Michael

albert21martins, 2018-04-12:
Hi I have also a PX4 2.1 and a Here with this compass. Is there a pre-release where we could work and integrate in the release 1.7.3 stable?

LorenzMeier, 2018-05-27:
Closing as duplicate.

closed 2018-05-27 by LorenzMeier