PX4/PX4-Autopilot #8628: [Multicopter mixer] Wrongly computed mixing tables

bresch, 2018-01-08:
I was checking the new [mixer generator](https://github.com/PX4/Firmware/blob/master/src/lib/mixer/geometries/tools/px_generate_mixers.py) script from @jlecoeur  in order to add my specific multicopter geometry and I found that it produces values that didn't match my mixing table generated by a custom script.
After checking the generator, I found that the table generated doesn't match the mixing strategy used in PX4.
The table generated has the following coefficients for each actuator:
```
roll_gain | pitch_gain | yaw_gain | thrust_gain
``` 
And has to be used like this: 
`out = roll_gain*roll + pitch_gain*pitch + yaw_gain*yaw + thrust_gain*thrust` (which is the standard matrix*vector multiplication)
However, in PX4, the equation is 
`out = (roll_gain*roll + pitch_gain*pitch + yaw_gain*yaw + thrust)*out_gain` 
https://github.com/PX4/Firmware/blob/26d95ef67479b90212f5ea8255d43721865ff904/src/lib/mixer/mixer_multirotor.cpp#L188-L192

At the current state, there is a confusion and the `thrust_gain` generated in the tables is used as the `out_gain` which lead to an incorrect mixing and produces unexpected coupling between the axes.

This can be corrected in two different ways:
1. Fix the generator script to match firmware implementation
2. Change the PX4 implementation to use a simple matrix-style mixer

Since the gain of each motor can be chosen in the geometry config file, and that the matrix-style mixing table is simpler and more intuitive, I would prefer to use the 2nd alternative.
FYI @MaEtUgR @jlecoeur

MaEtUgR, 2018-01-08:
Seems like a nice catch...
I personally vote for the "matrix-style mixer" solution 2 because I think it's a lot more common and probably more intuitive.

jlecoeur, 2018-01-08:
I agree, let's fix mixer_multirotor.cpp to use
`out = roll_gain*roll + pitch_gain*pitch + yaw_gain*yaw + thrust_gain*thrust`

On what platform was the issue noticeable ?

jlecoeur, 2018-01-08:
It seems like the only affected platforms are those where gain_thrust is not the same for all motors. That is quad_deadcat, quad_vtail, hex_t and octa_cox_wide.

bresch, 2018-01-08:
@jlecoeur It is also noticeable when the platform is not symmetric. My hex is like this:
``` 
 (3)           (5)   ^
  |             |    |
  |             |    b
 (2)           (1)^  |
  |_____________| a  v
  |             | x
  |             | |
  |             | c
  |             | |
  |             | |
 (6)      <-d->(4)v
```

jlecoeur, 2018-01-08:
@bresch Can you send your geometry (toml) file?

bresch, 2018-01-08:
@jlecoeur here it is:
```
[rotor_default]
axis      = [0.0, 0.0, -1.0]
Ct        = 1.0
Cm        = 0.05

[[rotors]]
name      = "mid_right"
position  = [0.2375, 0.71875, 1.0]
direction = "CW"

[[rotors]]
name      = "mid_left"
position  = [0.2375, -0.71875, 1.0]
direction = "CCW"

[[rotors]]
name      = "front_left"
position  = [0.95, -0.71875, -1.0]
direction = "CW"

[[rotors]]
name      = "rear_right"
position  = [-1.0, 0.71875, -1.0]
direction = "CCW"

[[rotors]]
name      = "front_right"
position  = [0.95, 0.71875, -1.0]
direction = "CCW"

[[rotors]]
name      = "rear_left"
position  = [-1.0, -0.71875, -1.0]
direction = "CW"
```

jlecoeur, 2018-01-08:
Yep, your two rear motors are more loaded:
```cpp
const MultirotorMixer::Rotor _config_hex_h[] = {
	{ -1.000000,  0.129193, -1.000000,  0.891898 },
	{  1.000000,  0.129193,  1.000000,  0.891898 },
	{  0.500000,  0.655192, -0.500000,  0.829657 },
	{ -0.500000, -0.784385,  0.500000,  1.000000 },
	{ -0.500000,  0.655192,  0.500000,  0.829657 },
	{  0.500000, -0.784385, -0.500000,  1.000000 },
};
```
If you can move the CG forward such that c == a + b, that should fix the coupling issue until a software fix is implemented.

bresch, 2018-01-08:
@jlecoeur I could, but since it's a VTOL the CG is much more important for the FW static stability. I can still fly with my custom mixer script, nothing urgent for me; I just wanted to compare the values with the new script :)

PR #8639 (2018-01-17): MultirotorMixer: Apply thrust gain independently from other axes
Align the way motor output is computed with the model used in the mixer generation script.
previous:
`out = (roll_gain*roll + pitch_gain*pitch + yaw_gain*yaw + thrust)*out_gain`
new:
`out = roll_gain*roll + pitch_gain*pitch + yaw_gain*yaw + thrust_gain*thrust`
which is the standard matrix*vector multiplication.

This commit has 0 effect on symmetric platforms (i.e. most) as thrust_gain==1 on these platforms.
On asymmetric platform --where some thrust_gain are lower than 1-- the previous formulation resulted in coupling between th [... cut]
commit 262d9c79 (2018-01-17): MultirotorMixer: Apply thrust gain independently from other axes

Align the way motor output is computed with the model used in the mixer generation script.
previous:
`out = (roll_gain*roll + pitch_gain*pitch + yaw_gain*yaw + thrust)*out_gain`
new:
`out = roll_gain*roll + pitch_gain*pitch + yaw_gain*yaw + thrust_gain*thrust`
which is the standard matrix*vector multiplication.

This commit has 0 effect on symmetric platforms (i.e. most) as thrust_gain==1 on these platforms.
On asymmetric platform --where some thrust_gain are lower than 1-- the previous formulation resulted in coupling between t [... cut]
closed 2018-01-17 by dagar