PX4/PX4-Autopilot #8632: Possible to Arm wo GPS Lock

tops4u, 2018-01-08:
**Bug Report**

I could Arm in the Basement with an attached GPS (which did not get a valid Lock) but all Parameters set to - what I assumed - Values preventing Arming when GPS Missing or no valid Lock is available. 

CBRK_GPS_FAIL = 0
COM_ARM_WO_GPS = Disabled
EKF2_GPS_CHECK = 21
EKF2_REQ_EPH = 5.0m
EKF2_REQ_EPV = 8.0m
EKF2_REQ_NSATS = 6
EKF2_REQ_GDOP = 2.5

There was an Error Indication though : `Preflight fail: EKF not using GPS`

Check Log: https://logs.px4.io/plot_app?log=2ad190a6-08df-40bd-96d8-a423fc393970

tops4u, 2018-01-08:
This is not a duplicate to #8269 since here it is always possible to arm.

dagar, 2018-01-10:
I think you've got an important one here. These checks are no longer blocking arming.

tops4u, 2018-05-03:
Had a flight today where I could ARM but did get Position Mode Rejection because of large horizontal Error. Despite COM_ARM_EKF_POS = 0.5m / Log shows > 40m at the beginning. 

So I could only fly in Alt Mode, despite the OSD showed around 13 available Sats, obviously the horizontal Error was still very large. Is there any Progress on this one?

dagar, 2018-05-03:
Which version of firmware? Can you share the log?

tops4u, 2018-05-03:
Yes of course : https://review.px4.io/plot_app?log=c902429b-8467-42fb-996c-7bb2239ce229

dagar, 2018-05-03:
This has been fixed in https://github.com/PX4/Firmware/pull/8771

tops4u, 2018-05-03:
Ok. I‘m going to try Master then.

closed 2018-05-03 by tops4u