PX4/PX4-Autopilot #8634: PX4Flow with EKF2

yousafbinfarrukh, 2018-01-09:
Hey all, 

I'm trying to implement the ekf2 with px4flow but I cannot switch to the position hold. Moreover, when I see the lcoal position topics, the values are always zero. I've set the EKF2 aid mask to use optical flow, enabled the Lidar Lite and set primary height sensor as range finder. The Lidar is working right, I can see the ground clearance topic changing as I move the quadcopter with my hands. Moreover, the quality I am getting is arround 240+.
Thanks in advance :)

AlexisTM, 2018-01-09:
Can you check EKF2_AID_..  and enable PX4Flow fusion?

yousafbinfarrukh, 2018-01-09:
I've already done that

mzahana, 2018-01-10:
@yousafbinfarrukh I am currently using Px4Flow module with EKF2.
* I updated the firmware of px4flow using QGC
* I used an older QGC I guess 2.9... to connect QGC to px4flow directly  using USB in order to setup the camera focus. The recent version does not connect to pxx4flow.

I can see the Flow topic through the topic analyzer in QGC (switched back to the recent QGC version). I can also do position hold. However, the issue now is that the quadcopter drift significantly with time. I am not sure what needs to be tuned to get it to hold position reasonably well.

yousafbinfarrukh, 2018-01-13:
@mzahana can you plz tell me all the parameters you changed to get the optical flow working? How can I see if the flow is fused?

mzahana, 2018-01-13:
* updated PX4Flow firmware using QGC
* calibrated the lens focus as discussed [here](https://dev.px4.io/en/tutorials/optical_flow.html). You will need an older version of QGC to connect PX4FLOW to QGC via USB. I used 2.9.
* For the fight controller, I chose EKF2 as the estimator
* modified the `EKF2_AID_MASK` to disable GPS and enable optical flow
* I had a Lidar Lite v3, so I used it as the primary source for height
* connect PX4FLOW to FCU using I2C cable
* run the FCU, and check that the optical flow MAVLink packet is received. In QGC, go to the *Widgets* menu and open the MAVLink inspector. Check that the optical flow packet is being received.
* If you are able to switch to position hold mode, then the PX4Flow is being fused.

However, that does not mean that you will get good position hold. I am able to get an OK position hold outdoors. Indoors, I am getting a lot of drift in position hold mode, and have not figured out how to improve it.

Hope it helps.

tingtingxing, 2018-10-20:
Hello all,
I have the similar problem with @yousafbinfarrukh and I have done all of the steps like @mzahana said here. I connected the PX4Flow with Pixhawk via I2C for positioning in door.
As I can see in the Mavlink Inspector in QGC, the Flow message (OPTICAL_FLOW_RAD) can be received but the local position is always 0 (both LOCAL_POSITION_NED_x and LOCAL_POSITION_NED_y). 
It seems that local position is not valid?
Do you know why this problem occurs, or where can I find the reasons for local position is not valid in Firmware code?
I really appreciate for your help.
please help me.

AlexisTM, 2018-10-21:
You've to enable the fusion parameter and ensure the flow quality is high (~200+)

tingtingxing, 2018-10-21:
@AlexisTM 
Thank you so much about the reply. The flow quality of my PX4Flow was lower than 200 before. 
After changing the surface, to which the PX4Flow faces, now the flow quality is higher than 200 during the test.

About the fusion parameter:
EKF2_AID_MASK was set to 2
SYS_MC_EST_GROUP was set to 2

But the LOCAL_POSITION_NED.x and -.y are still keeping zero.
Did I forgotten to set some other parameters?

There are three pictures about the test of my PX4Flow:
[image]

[image]
[image]

I really appreciate for help in advance!

Ken-Thorburn, 2019-10-21:
@tingtingxing Hey, this is a problem I've been dealing with as well. I eventually gave up on EKF2 and switched to LPE as I found no way to get it working.

gouyiyong, 2019-10-30:
> @AlexisTM
> Thank you so much about the reply. The flow quality of my PX4Flow was lower than 200 before.
> After changing the surface, to which the PX4Flow faces, now the flow quality is higher than 200 [...]

@tingtingxing  hi, do you solve this issue, now I have the same issue.