PX4/PX4-Autopilot #8643: mRo pixhawk detect bad HW version after reboot

svpcom, 2018-01-09:
After reboot mRo pixhawk (https://docs.px4.io/en/flight_controller/mro_pixhawk.html)
changes HW version from 0x8 to 0xB which leads to L3GD20 and LSM303D detection fails.
If I've manually override HW detection to 0x8 it works. Other pixhawk (v2.4.6 from aliexpress) works well.

HW and SW info:
```
nsh> ver all
HW arch: PX4FMU_V2
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: 50bd148f53f1aeca6f1bd4a1caabcc18e4f2888b
FW version: 1.7.3 0 (17236736)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: b18053574bf41712cef93e31bf178518f451a350
Build datetime: Jan  9 2018 16:19:06
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
MFGUID: 313637323037510e0033003a
MCU: STM32F42x, rev. 3
UID: 33003A:3037510E:31363732 
```

Boot logs:
``` .diff
$ diff -u /tmp/good.log /tmp/bad.log 
--- /tmp/good.log	2018-01-10 01:04:22.353481536 +0300
+++ /tmp/bad.log	2018-01-10 01:03:50.586355492 +0300
@@ -1,11 +1,11 @@
-INFO  [drivers__boards__px4fmu-v2] Ver 0x8 : Rev 0 V2
+INFO  [drivers__boards__px4fmu-v2] Ver 0xB : Rev 0 V2
 [boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
 [boot] Fault Log is Armed
 sercon: Registering CDC/ACM serial driver
 sercon: Successfully registered the CDC/ACM serial driver
 HW arch: PX4FMU_V2
 HW type: V2
-HW version: 0x00090008
+HW version: 0x0009000B
 HW revision: 0x00000000
 FW git-hash: 50bd148f53f1aeca6f1bd4a1caabcc18e4f2888b
 FW version: 1.7.3 0 (17236736)
@@ -26,7 +26,7 @@
 WARN  [ms5611] no device on bus 4
 MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
 WARN  [bst] no devices found
-HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
+WARN  [hmc5883] no device on bus 1 (type: 2)
 WARN  [lis3mdl] no device on bus 2
 WARN  [hmc5883] no device on bus 2 (type: 1)
 WARN  [mpu6000] no device on bus #3 (SPI1)
@@ -38,9 +38,11 @@
 WARN  [mpu9250] no device on bus 3
 WARN  [mpu9250] probe failed! 255
 WARN  [mpu9250] no device on bus 4
-L3GD20 on SPI bus 1 at 1 (11000 KHz)
-LSM303D on SPI bus 1 at 2 (11000 KHz)
-Airspeed on I2C bus 1 at 0x21 (bus: 100 KHz, max: 100 KHz)
+ERROR [l3gd20] driver start failed
+WARN  [lsm303d] SPI init failed
+ERROR [lsm303d] driver start failed
+WARN  [sdp3x_airspeed] not started on bus 1
+WARN  [sdp3x_airspeed] not started on bus 2
 WARN  [ms5525_airspeed] not started on bus 1
 WARN  [ms4525_airspeed] not started on bus 1
 WARN  [ms4525_airspeed] not started on bus 2
@@ -50,7 +52,7 @@
 INFO  [px4io] default PWM output device
 INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
 INFO  [mavlink] mode: Onboard, data rate: 10000 B/s on /dev/ttyS2 @ 1500000B
```

bkueng, 2018-01-10:
I can reproduce this on my mRo Pixhawk. @davids5  any ideas?

davids5, 2018-01-10:
My guess is that some driver is changing a GPIO to state where it is different from power on reset. On the reboot it is affecting the detection. IIRC, The detection is based on sensing no-connects. I wI'll need to look at it on the bench when I can.

svpcom, 2018-01-10:
@davids5 Why detection is done by some GPIO sensing magic? May be board ID should be stored in flash (for example in bootloader area)?

LorenzMeier, 2018-01-10:
@svpcom Because manufacturers did deviate from what was the reference standard that is required to operate correctly.

This is something we will likely tolerate less in the future.

davids5, 2018-01-10:
@svpcom

The root cause of the issue is non compatible Hardware using the same bootloader and Reporting that it's FMUv2. 

The reason it was done this way was to aviod requiring a bootloader update. But I think that's going to be an inevitable.

davids5, 2018-01-10:
@svpcom - yes if we alter the bootloader OR just change the non compatible Hardware's bootloader to use a different ID the detection and be build OR run time. FMUv5 solves this with 10 HW Revision and version ID code points.

orlando3d, 2018-01-16:
@davids5 
Hi , I can confirm that mRo pixhawk hardware adheres to the official hardware v2.4.6 and also that this behavior "HW version: 0x0009000B after softreset" happens in both 3DR and mRo units except in my "HK/Bangood" HW version on hand.

What I found differently in this "HK/Bangood" version on hand is that both R603 and R635 are populated and only one can be populated as stated in the schematic for proper sensors_VDD controlling

 
[image]

I'll try to dig in more..

svpcom, 2018-01-16:
@davids5 @LorenzMeier 
I've found a root of problem: Commit 75b93b0728d57c642641c354dc99390a6ac43ee8 was not properly reverted. I've created a pull-request:
https://github.com/PX4/Firmware/pull/8703

svpcom, 2018-01-16:
Affected all stable versions from v1.7.1 and newer

LorenzMeier, 2018-01-16:
The intent was to not revert it entirely because hundreds of boards (and hence users) would then be unable to boot.

svpcom, 2018-01-16:
But without this all other pixhawk boards (for example my from mRo, banggood and aliexpress) lost all sensors after reboot.

svpcom, 2018-01-16:
@LorenzMeier What was the HW problem was with HK boards that require this fix?

LorenzMeier, 2018-01-16:
As I said, we will have to sort this out properly by testing in parallel on all affected boards. @dagar and I have them now, but we will only merge changes that satisfy all of them. Otherwise we could handle a reboot on the mRo boards differently.

orlando3d, 2018-01-16:
This may help regarding HK fix:
After reviewing one of my Generic Bangood Pixhawk that was affected by the non-booting issue, I found a design deviation where both R635 and R603 are populated and by design only one should be populated for proper SENSORS_VDD controlling.

[image]

svpcom, 2018-01-16:
But how it works before HK fix?

dagar, 2018-01-17:
I'm getting everything together to do some bulk testing tomorrow.

svpcom, 2018-02-01:
@dagar Any updates?

dagar, 2018-02-11:
Testing notes here. https://github.com/PX4/Firmware/pull/8703#issuecomment-364713594

dagar, 2018-03-14:
Please see https://github.com/PX4/Firmware/pull/9074

LorenzMeier, 2018-03-14:
I'm closing this - please report back if the PR didn't fix it for you.

closed 2018-03-14 by LorenzMeier