PX4/PX4-Autopilot #8644: Sensor module validator bug?

lijieamd, 2018-01-10:
**Bug Report**

I'm a little confused with some sensor module and ecl validation source code

https://github.com/PX4/Firmware/blob/0ead55d38afb5b4c30b33c839f079af5dfd731e1/src/modules/sensors/voted_sensors_update.cpp#L793-L794

here put sensor data to validator are using **uorb_index** from 0~n(multi sensor)

but look at the validator

https://github.com/PX4/ecl/blob/bdbc05cf91e6b13e2033ee3a7ecd3bdbf6f22f09/validation/data_validator_group.cpp#L72-L81

it's a singly linked list to group multi sensor, the **_first** is actually the last one we add to group, so i imagine a situation:
    there are 2 mag sensor, publish *sensor_mag0* and *sensor_mag1* orb messages, then sensor module subscribe to *sensor_mag0* first, at this moment everything is ok, validator group has just one node, **uorb_index = 0** is it, but later sensor module subscribe to *sensor_mag1*, at this time the **uorb_index = 0(mag0)** data is actually put to **_first** in group which is **mag1**, and mess the validator internal state.
    
so it's my misunderstand or a bug?

bkueng, 2018-01-10:
You are correct. Nice find!
`add_new_validator()` should add a new item to the end of the linked list. Do you want to create a PR for that?

lijieamd, 2018-01-10:
I'm not sure if we should treat it as a Firmware bug or a ecl lib bug?
if no other modules also use ecl validation lib or rely on the fact that ***_first*** is actually the last node, i think i can fix it in ecl repository

bkueng, 2018-01-10:
I think there are no other users of that ecl functionality, but @priseborough knows better.

priseborough, 2018-01-10:
It's not used anywhere else so it makes sense to fix it in ecl.

lijieamd, 2018-01-11:
Ok, i'll create a PR in ecl repository