PX4/PX4-Autopilot #8648: Multi-Vehicle question.

CrazyDominic, 2018-01-10:
Finally, I can see two drones on gazebo used Multi-Vehicle.

But, I can only on drone control.

How to control another one?

hamishwillee, 2018-01-10:
Are you connected via QGC? https://docs.qgroundcontrol.com/en/releases/stable_v3.2_long.html#multi-vehicle-view

CrazyDominic, 2018-01-10:
um... no not this mode. If I execute QGC, just first drone connected.

hamishwillee, 2018-01-10:
I haven't done it other than with mock links. I imagine that you'll have to manually connect the second drone because QGC is only looking for one UDP port for auto connection. 

The right place for this sort of question is here: http://discuss.px4.io/c/qgroundcontrol/qgroundcontrol-usage

CrazyDominic, 2018-01-10:
This is my situation, I want to control my second drone not just first.

Is there another way to connect not a QGC? like MAVproxy.

[image]

hamishwillee, 2018-01-10:
I have no idea. But even if you wanted to use mavproxy you'd still need to know the UDP connection of the second drone ... at which point you might as well use QGC.

mzahana, 2018-01-10:
@CrazyDominic I was involved in the mutli-vehicle setup interface. How are you running the setup? I was mainly targeting ROS based interface. So I was relaying to QGC through MAVROS. Also, do you have unique `MAVLINK_SYS_ID` for each vehicle that is defined in the SITL parameter files, one for each vehicle?
Have you looked at the dev guide at https://dev.px4.io/en/simulation/multi-vehicle-simulation.html ?

CrazyDominic, 2018-01-10:
@mzahana 
Yes, I follow this guide. So, I can see two drones on my simulator and QGC. Each drone can control seperatly. But, cannot be controlled QGC simultenously. I want to controll each drone using each command. 

In QGC, I command to vehicle 1 to execute mission, and I command to vehicle 2 to execute other mission continuously. Therefore vehicle 1 stop mission, and only vehicle 2 execute that.

hamishwillee, 2018-01-10:
Ah, so what you're saying is that you want to be able to send a single command to both drones to do the same thing? ie. you send both drones the command to takeoff, and then both to follow a mission?

If so, that is not supported in QGC. You could probably write a [dronecore](http://dronecore.io/) app to do that.

CrazyDominic, 2018-01-12:
@hamishwillee I really thank you. I need to study more...

closed 2018-01-12 by CrazyDominic

DarshanSavaliyaOfficial, 2020-09-05:
Hii, I am having a similar issue, I tried to connect two drones with Dronekit to QGC using mavproxy. The terminal shows that both the vehicles are connected but I can't see the second drone in QGC. How to solve this issue?
Thank you

Seunghwan17, 2020-09-06:
@DarshanSavaliyaOfficial HI. CrazyDominic is my previous nickname. :)
So, which drones connected to QGC? Are drones SITL or real drones? If there are SITL, you need to set-up difference ID for each other.

DarshanSavaliyaOfficial, 2020-12-28:
yes @Seunghwan17, thank you for your support. I had to set-up different IDs for both and had to run one drone in the background. This worked for me.