PX4/PX4-Autopilot #8650: Support for SRXL receiver (e.g. spm4649t)

taileron, 2018-01-10:
this protocol would allow to send telemetry information from the fc back to the tx and is extremely lightweight, are there plans to support it in future?

dagar, 2018-01-10:
It definitely looks possible. I see a basic specification for SRXL, but so far haven't found anything about the actual telemetry fields. Do you know if they're documented?

taileron, 2018-01-11:
There are two documents available:
https://www.horizonhobby.com/pdf/SPM4649T_Bi-Directional_SRXL_specifications.pdf
https://www.spektrumrc.com/ProdInfo/Files/SPM_Telemetry_Developers_Specs.pdf

dagar, 2018-01-11:
Oh ok, so the specific messages are X-Bus. Do you have one of these receivers? Are you interested in developing any of it yourself?

CarlOlsson, 2018-01-11:
SRXL is also the protocol for Multiplex receivers
[SRXL-MULTIPLEX_V2.pdf](https://github.com/PX4/Firmware/files/1624003/SRXL-MULTIPLEX_V2.pdf)

Would be awesome to add support for it since you currently need to use a PWM to PPM converter to use Multiplex receivers.

taileron, 2018-01-16:
I´m using this receivers (SPM4649T) successful with Betaflight. Without serial telemetry and PX4 there is an additional a/d input for sending a lipo voltage alarm telemetry back to the tx. I´m not enough skilled in coding for developing in such a project ....

pedrodparkes, 2018-03-04:
hey, I use JR X-BUS B (that is SRXL). It works fine on Spektrum port with ArduPlane firmware.
How can I make px4 firmware work with it?

Able to build/develop/run test-flights.

dagar, 2018-03-05:
@pedrodparkes that's great. We'd need a decoder in the RC lib. https://github.com/PX4/Firmware/tree/master/src/lib/rc

Then with that in place it needs to be connected in a few places (px4fmu, px4io, etc). Example https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp#L1571

kekiefer, 2018-06-09:
Just a heads up that I've been poking at this over the last couple weeks and have managed to get some simulated flight pack data back to the transmitter by reconfiguring one of the telemetry UARTs. Critically, the SRXL UART needs to be in half duplex mode, and on the STM32 can only be done on a UART TX pin, not an RX pin like the one that is wired for DSM on every platform. I'm doing this work on a Pixhawk 4, and unfortunately schematics aren't available so I'm not sure if there is another possibility on that platform. It looks like there might a resistor addition possible on the Pixracer in order to use the DSM port, and I haven't really looked at others. Maybe someone with access to the Pixhawk 4 schematics could chime in here.

kekiefer, 2018-07-22:
RSSI telemetry works from these receivers now. #9998 is outstanding to make it work also on boards with PX4IO.

Also with #9898 merged (improved single-wire support) the road is clear for the rest of the telemetry integration. Data is going back to the RC properly so now it's just a matter of cleaning up the code for submission.

pedrodparkes, 2018-08-02:
what's the state for JR xBUS?
is it still not working?

kekiefer, 2018-08-03:
I'm not inventing any generic srxl logic to handle this, so outside the scope of my work. Keeping changes to a minimum, so I won't change the receiver logic at all beyond echo suppression of the half-duplex line, and the transmitter logic will be rather spektrum specific.

closed 2019-02-11 by stale[bot]