PX4/PX4-Autopilot #8654: SDLOG: Time Stamp totally wrong

tubeme, 2018-01-11:
Despite many attempts to fix this problem it is again persistent.

For two files:

First File

Real time: 7:48 
Log File Time: 15:48 (It is UTC time obviously)
In review.px4.io report Logging Start: 17:46

Second File

Real Time: 8:09
Log File Time: 16:09 (UTC)
In review.px4.io report Logging Start: 18:06

Strange...

We have preset

SDLOG_UTC_OFFSET = -480 (We are GMT-8:00)

tops4u, 2018-01-11:
I have an explanation but not a fix for this. 

The Development of the Web Log Viewer was made in TC UTC+2 (for ex. Switzerland, no DST - current). This would explain the major offset of 10h (your UTC -8 plus Switzerland UTC +2). 

As to the minor Offset, I have noticed that it seems to use the Boottime as a basis to Log. When you have multiple consecutive Flights 0:00 will always be Boot-Time so all the Flights will get Timestamps after each other (while powered). 

As for the general Timezone Issue, this should maybe go here: https://github.com/PX4/flight_review

tubeme, 2018-01-11:
hy then it can get the date from the GPS lock and create folder with the date and it is not possible to have a correct time stamp based on GPS Lock + time zone offset?

It is very inconvenient because we are working on a couple of locations in different zones and record a video with correct time stamp. In order to sync the video to the LOG files it is a nightmare to make this calculation every time manually...

bkueng, 2018-01-11:
- `SDLOG_UTC_OFFSET` is used for the log file name. So changing that should give you a different file name. Does that work? (or what exactly are you referring to as 'Log File Time'?)
- Flight Review: it uses the UTC timestamp [from the GPS](https://github.com/PX4/flight_review/blob/master/plot_app/plotted_tables.py#L118), and then converts it [here](https://github.com/PX4/flight_review/blob/master/plot_app/templates/main.js#L38) to the browsers local timezone. There are a bit [different and conflicting](https://stackoverflow.com/questions/13622142/javascript-to-convert-utc-to-local-time) methods to do that, but it should be correct as is. What browser do you use?

Can you provide the logs?

tubeme, 2018-01-11:
Sorry Log File Time stamp i meant... Not the name but the time stamp of the log file...

https://logs.px4.io/plot_app?log=a11fdda4-1d23-48c2-8596-f5c5a16e23ba

https://logs.px4.io/plot_app?log=0963d4d3-c68a-4a31-9011-3c6292fb879c

tubeme, 2018-01-11:
@bkueng Do you need the original files?