PX4/PX4-Autopilot #8655: VTOL: Back transition after the end Nasty Pitch up.

tubeme, 2018-01-11:
Hi guys,

We have a strange problem with the back transition with our delta-quad.

The problem is only in Position mode. No such problem when in Altitude or Stabalized modes.

After the back transition is finished the MC sharply makes a putch up command.

We have MPC_DEC_HOR_MAX and MPC_ACC_HOR_MAX set to 2 but still this kick back.

[image]

Here is the log:
https://logs.px4.io/plot_app?log=a11fdda4-1d23-48c2-8596-f5c5a16e23ba

Any Ideas?

dagar, 2018-01-11:
Any difference in mission mode?

tubeme, 2018-01-11:
Mission mode is the same! So probably it is in the Position controller.

tubeme, 2018-01-11:
Also wonder why there is no Pitch nor Roll Set Points during the transition.

We fly without airspeed sensor!

dagar, 2018-01-11:
I believe it's the way the multicopter position controller is reinitialized and seeded during transition. As a temporary improvement you can reduce MC_PITCHRATE_MAX.

sanderux, 2018-01-11:
This is because your velocity when coming out of backtransition is still 6m/s while your mpc_xy_vel_max is set to 4. Increase your back transition duration or use reverse thrust to brake actively

closed 2018-01-11 by sanderux

tubeme, 2018-01-12:
@sanderux Thanks a lot!