PX4/PX4-Autopilot #8656: EKF2-PX4FLOW: Significant drift in position hold mode

mzahana, 2018-01-11:
Hi.

My quadrotor drifts significantly in position hold mode.
* Firmware version:  stable v1.7.1
* FCU: MindPX
* Using Lidar Lite v3. Works well for altitude stabilization
* EKF2 estimator. Disabled GPS in `EKF2_AID_MASK` and enabled optical flow fusion
* `EKF2_FLOW_DELAY` and other flow related parameters are left at default values.

Once switched to position hold mode, it stays in position for a second, then starts to drift around. Within few seconds, it can be couple of meters away from the set position.

I am not sure what the problem could be. So, here is a log.
https://logs.px4.io/plot_app?log=b4e0b04e-c7f3-4039-ba37-4845d43bafd0

Any hints are appreciated. Thanks.

mzahana, 2018-01-13:
@ChristophTobler Could you please check this?

ChristophTobler, 2018-01-13:
@mzahana Sure:+1: I'll have a look

ChristophTobler, 2018-01-13:
It looks like there is a bit more vibration on the FCU than normal but I don't think that this causes the problems you're describing...
Have you been able to fly this quad with GPS?
Did you check the orientation of the PX4Flow? The pixel flow output should be as follows
moving forwards: +y
moving backwards: -y
moving right: -x
moving left: +x

You can check that in QGroundControl->Analyze

mzahana, 2018-01-13:
I flew outdoors many times with GPS and it works very well in position
hold. I also flew it with mocap setup and works very well. Today, I also
flew it outdoors with px4flow (I disabled GPS in EKF2_AID_MASK and enabled
flow), and it did way better in position hold compared to indoors.

<https://github.com/PX4/Firmware/issues/8656#issuecomment-357459416>, or mute
the thread

mzahana, 2018-01-13:
Could it be the floor texture? I have a black foam floor with no rich features.

ChristophTobler, 2018-01-13:
That could certainly be the case. Can you try again on something else e.g. carpet?

tubeme, 2018-01-14:
I'm not exactly sure about the flow sensor, but given it is a camera there are two very important components in order to be precise. 1. Enough Light, 2. A good contrasting pattern in the frame of the image it picks up. 3. You should make sure there is no vibration. Vibrations kill the image especially if you need to measure movement in it.

ChristophTobler, 2018-02-14:
@mzahana Please reopen if still an issue

closed 2018-02-14 by ChristophTobler