PX4/PX4-Autopilot #8657: Snapdragon: params don't get synced to DSP when specified without qshell

darioxz, 2018-01-11:
I would like to set the `EKF2_AID_MASK` and `EKF2_HGT_MODE` parameters in `px4.config`:

```
uorb start
muorb start
logger start -t -b 200
param set MAV_BROADCAST 1
#param set SDLOG_PRIO_BOOST 3
# EKF2: fuse external vision pose estimate
param set EKF2_AID_MASK 24
# EKF2L Switch from barometer to vision as main source for height estimation
param set EKF2_HGT_MODE 3
#param set EKF2_EVP_NOISE 0.01
#param set EKF2_EVA_NOISE 0.01
dataman start
navigator start
mavlink start -u 14556 -r 1000000 -t 10.20.21.138
sleep 1
mavlink stream -u 14556 -s HIGHRES_IMU -r 10
mavlink stream -u 14556 -s ATTITUDE -r 10
mavlink stream -u 14556 -s RC_CHANNELS -r 10
```

The correct parameter is shown after executing `pxh> param show EKF2_AID_MASK`. Unfortunately, the parameter doesn't get synced to the DSP part: `qshell param show EKF2_AID_MASK` outputs the old value, not the value specified in `px4.config`. 

`pxh> param set EKF2_AID_MASK 24`  doesn't sync the value either - only `pxh> qshell param set EKF2_AID_MASK 24` syncs the value correctly.

Is this the intended behavior or is it a bug?

LorenzMeier, 2018-01-13:
@darioxz Definitely a bug. Can you try again on current master? We fixed params on Aero.

ChristophTobler, 2018-01-15:
Still exists on Snapdragon

ChristophTobler, 2018-01-15:
Fix in https://github.com/PX4/Firmware/pull/8686. @darioxz can you please give this a try?

darioxz, 2018-01-15:
#8686 fixed the issue, thanks a lot!

closed 2018-01-15 by ChristophTobler