PX4/PX4-Autopilot #8664: VTOL: Front transition Ramp Up

tubeme, 2018-01-12:
@Daniel and @sanderux 

Because we fly without airspeed sensor, during the front transition everything is timing. So in Position / Auto modes the the ramp up of the front transition is something like this:

[image]

Visible from the graph:

[image]

[image]

In the Log:

https://logs.px4.io/plot_app?log=4d974849-6f42-4b03-945d-8e0eb96e2f29

I could not find a way to make the ramp up of the pusher motor like this during the transition:

[image]

I found a parameter for the back transition ramp up:

VT_B_TRANS_RAMP which is the reverse to what I want to achieve.

Am I missing something?

Can I use VT_TRANS_P2_DUR to make it 1-2 seconds to achieve this?

tubeme, 2018-01-13:
May be is not a bad idea to have a VT_F_TRANS_RAMP parameter for the front transition ramp up as well.

tubeme, 2018-01-14:
This is not a proper ramp up for Front transition for VTOL in Position and Mission Modes.

[image]

This is like having the real plane ramp up from 0 to full throttle from the beginning of the runway to the end of it.

With VT_TRANS_THR=0.7 every time after the transition the plane is almost at stall. It is seen from the throttle demand after the transition.

tubeme, 2018-01-31:
@sanderux Any ideas about the front ramp up? I could not find a way to control it.

sanderux, 2018-02-03:
you need to set VT_F_TRANS_DUR to 1 (thats the ramp-up time for front transition)

sanderux, 2018-02-03:
also 5 seconds is probably too little for a deltaquad

closed 2018-02-03 by sanderux

tubeme, 2018-02-06:
Strange... 
Here is from parameters reference...

VT_F_TRANS_DUR
(FLOAT) | 
Duration of a front transition
Comment: 
Time in seconds used for a transition

Thanks for the info I will note it.