PX4/PX4-Autopilot #8674: MC: Almost crash, about 1 sec strange PosCTL Setpoint behaviour

tops4u, 2018-01-13:
**Bug Report**

Since changing ESC for some more aggressive Variant (BLHeli Damped Light replaced "conventional" ESC), I have a Problem when Braking in PosCTL Mode I get a strong Oscilation (probably by switching to Hold/Loiter when Stick is completely centered). I tried to figure MPC Settings in order to sort this out. I have still some strong Oscillation even when slowly moving the Stick back to center. 

Tried in about 12m Alt to check some Roll Movements. Just after centering Roll Stick from a Left Bank, the Copter started to get uncontrollable and spiraled down about halfway when it started to fight the motion and managed to just stabilize about 1-2m off the Ground. 

I would like some Advice on this as to:
- How should I tune MPC Params in order to avoid the Oscillation when braking in PosCTL?
- Why were the Setpoints of Roll/Pitch kind of "sleeping from 3:57 to 3:58.5? There it looked like the Copter got out of control and started to stabilize only after 3:58.5. Also Check Main PWM for Ch0 & 1 it looks like they almost locked on min/max just to swap at about 3:58.5

Any hints are appreciated as I really start to loose trust in my Copter. 

Log is here: https://logs.px4.io/plot_app?log=391170a2-63e6-4f20-9e5b-14d9f33a67ae

LorenzMeier, 2018-01-13:
Still looking at it - but this jamming indicator "step function" is quite unusual. Is there any particular electrical equipment on the drone?

https://logs.px4.io/plot_app?log=391170a2-63e6-4f20-9e5b-14d9f33a67ae#Nav-GPS-Noise-_-Jamming

LorenzMeier, 2018-01-13:
After looking through the log a little further it looks like some type of almost or actual resonance: https://logs.px4.io/plot_app?log=391170a2-63e6-4f20-9e5b-14d9f33a67ae#Nav-Raw-Angular-Speed-Gyroscope

The frequency plot of the system is completely changing: https://logs.px4.io/plot_app?log=391170a2-63e6-4f20-9e5b-14d9f33a67ae#Nav-Acceleration-Power-Spectral-Density

I would cross-check that your frame is stiff enough and not resonating in the propeller frequency. And you probably need to lower the rate gains as you seem to get a stronger response now. I'm not sure yet what happened in that moment, but its possible the rotor blades stalled (due to the sudden speed changes) and stopped producing proper lift. That's speculation right now though, so don't take that as a final verdict.

tops4u, 2018-01-13:
Got a small Clip from the Onboard Cam. 

[AlmostCrash.zip](https://github.com/PX4/Firmware/files/1628488/AlmostCrash.zip)

tops4u, 2018-01-13:
I have lowered Pitchrate from the former 6.5 to 5. The Frame is a Tarot 680Pro and the Motors are TMotor 4010/475kV with TMotor 13x4.4. 

What I'm wondering is, what happend here:

[image]

and here:

[image]

While there was nothing going on, on the RC Side:

[image]

Nor anything too special in Gyro or Accel:

[image]

dagar, 2018-01-14:
I need to look closer tomorrow to see if this is cause or effect, but those attitude setpoint steps correspond with the multicopter position controller acceleration setpoint flatlining.

[image]

dagar, 2018-01-14:
The thrust setpoint is calculated here. https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2521

tops4u, 2018-01-14:
The Copter was flying with the identical Mechanical Setup (Motors, Props, Frame, etc) except for different ESC and different FC (with same Params) perfectly without any Problems.

tops4u, 2018-02-08:
Mechanical Setup is probably fine. After doing some work on the Copter I noticed, that one of the ESC Signal Groundcables was only loosely connected which may have caused the Starbord Motor to stop or turn on lower speed causing the Copter to dip over the righthand side with a clockwise spin. 

I'll close this Issue considering this an electrical failure.

closed 2018-02-08 by tops4u