PX4/PX4-Autopilot #8690: Preflight check airspeed driver grace period

potaito, 2018-01-15:
I am running into problems with the airspeed driver grace period during boot
https://github.com/PX4/Firmware/blob/18b83f2960879a71d19b456f5823d0abb2950290/src/modules/commander/PreflightCheck.cpp#L612

In my case the drone is booting too fast, meaning it runs the preflight check here 
https://github.com/PX4/Firmware/blob/18b83f2960879a71d19b456f5823d0abb2950290/src/modules/commander/commander.cpp#L1621 in under 2 seconds (the grace period), which then sets `status_flags.condition_system_sensors_initialized` to true. The preflight checks are repeated in the commander thread main loop, but the aforementioned variable is not corrected. 

I don't understand the commander that well and why that `2000000` period needs to be there in the first place, so I can't fix it. Any ideas how to solve this? I see two options:
1. Modify the preflight check so that the result is no longer binary, but can return a "not-ready" state. For instance it would return an enum with `SENSOR_FAIL, SENSORS_NOT_READY, SENSORS_OK`. However I don't think this is necessary. If sensors are not ready, the preflight check should fail and the failure should be handled by the caller. 
2. Modify commander so that  `status_flags.condition_system_sensors_initialized` is also updated in the commander's loop. Maybe this is against design rules, I don't know...

@dagar fyi

dagar, 2018-01-15:
I need to take a closer look, but I think what we need to do here is apply the grace period to the user facing error message, rather than the entire check. Arming would still be prevented until things are actually ready.

potaito, 2018-01-16:
That would make more sense :)

dagar, 2018-05-04:
This is resolved in master.

closed 2018-05-04 by dagar