PX4/PX4-Autopilot #8694: HIL GPS Triggers "Unknown Armed" Flight Mode

axdhill, 2018-01-15:
**Bug Report** 

I'm passing indoor position measurements via offboard computer using the telemetry port of my Pixhawk 2.1 running PX4 1.7.3 stable. The position measurements are streamed as fake GPS coordinates. The message types sent are:
`HEARTBEAT`
`LOCAL_POSITION_NED`
`HIL_GPS`
`GPS_INPUT`
The expected behavior is that px4 sees the incoming mavlink stream for indoor positioning (HIL GPS is enabled in PX4). 

In 1.7.3 stable, QGC sees the flight mode as flipping constantly from the correct mode (Stabilized, Unarmed) to "Unknown Armed." The copter is not actually armed. I can use the RC to really arm it and take off manually, and HOLD mode functions as expected (multicopter locks in position). However, I cannot use QGC to automatically takeoff - I assume because the flight mode is constantly flipping between Stabilized/Unknown and Armed/Unarmed.

In 1.6.5 this did not happen - the flight mode remained as expected even when GPS was fed in through telemetry.

axdhill, 2018-01-15:
[image]

santiago3dr, 2018-08-20:
this might be related to issue #8613

closed 2019-02-11 by stale[bot]