PX4/PX4-Autopilot #8696: VTOL: Reject position control / Hold / Mission after succeed GPS lock.

svpcom, 2018-01-15:
I've found a strange issue with v1.7.3: PX4 reject to switch to POSCTL, HOLD  or Mission.
GPS is running and got 3D lock. When I try to start mission from QGroundControl is shows message: "vehicle is not ready". I've tried to recalibrate all sensors, but it didn't help. All sensors status are good. 

https://logs.px4.io/plot_app?log=60f77638-9a48-4aac-95a8-66cf25eb19ae

dagar, 2018-01-15:
I'm not sure why yet, but you never had a global position estimate.

dagar, 2018-01-15:
[image]

svpcom, 2018-01-16:
I've got it. Yes i'm testing partially inside (on the balcony). But (IMHO) ground station should show more clear GPS status and provide messages why vehicle should not do some action. Because now most of users will panic in such case.

dagar, 2018-01-16:
I couldn't agree more. https://github.com/PX4/Firmware/issues/8697

RomanBapst, 2018-02-13:
@svpcom @dagar I'm going to close this because the main issue is addressed in #8697 
@svpcom If you want to give feedback as a user you can do that in #8697

closed 2018-02-13 by RomanBapst