PX4/PX4-Autopilot #8707: Parameter meta data: LPE_FUSION max value set above the value of an int32

DonLakeFlyer, 2018-01-16:
This causes QGC to spit out a warning about bad meta data. It also causes QGC unit tests to fail. On QGC side I hacked the value back to 255 which seems to match the bits used. But there must have been some reason for changing it in the firmware in the first place. So not sure what the correct value it.

dagar, 2018-01-17:
Looks like a bug to me. That's max UINT32
https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/params.c#L455

TSC21, 2018-01-17:
Yeah. To fix to UINT8MAX

DonLakeFlyer, 2018-01-17:
More new breaks:
```
    <parameter default="3" name="PLD_MAX_SRCH" type="INT32">
      <short_desc>Maximum number of search attempts</short_desc>
      <long_desc>Maximum number of times to seach for the landing target if it is lost during the precision landing.</long_desc>
      <min>0.0</min>
      <max>100</max>
      <scope>modules/navigator</scope>
    </parameter>
```

Float value specified for int param in min.

TSC21, 2018-01-17:
@DonLakeFlyer I will push a fix for both in a sec. Let me know if you find more in the mean time.

dagar, 2018-01-17:
Fixed by Nuno in https://github.com/PX4/Firmware/pull/8716

closed 2018-01-17 by dagar