PX4/PX4-Autopilot #8712: Multi-Plane SITL No Longer Works in Latest Stable Branch

mago3421, 2018-01-17:
[multi-plane-SITL-files.zip](https://github.com/PX4/Firmware/files/1639866/multi-plane-SITL-files.zip)
Hello,

My team uses the stable branch to test formation flight on live planes or in SITL. I recently checked out the latest stable so we could use RTPS instead of MAVLink but now the simulation does not work.

To get the simulation working, I clone a fresh copy of the stable branch. Afterwards, I run _make posix_sitl_default _gazebo_plane__ to build the firmware. This works and I can connect to the SITL using dronekit.

I then edit the _multi_uav_mavros_sitl.launch_ file to use planes which reads as follows:
[image]

**NOTE:** The XML for the second UAV is not shown but it is identical to the first UAV except for MAVLink ports, target system numbers, and rcS files. Beforehand, I clone the plane command file into two command files: plane_1 and plane_2. These command files are identical except for MAVLink streams, MAV SYS IDs, and  SITL_UDP_PORT param sets. I do the same for the plane sdf files.

After editing the multi-vehicle launch file, I edit the _single_vehcile_spawn.launch_ to read as so:
[image]

I comment out the xacro.py line because I am using sdf files for planes that I generated some time ago.

Next, I run the multi-vehicle simulation as shown in the next few screenshots below:
[image]
[image]
[image]

**NOTE:** I close the Gazebo GUI and the process finishes cleanly. I also get messages saying the uav-1_plane-1_spawn and uav-2_plane-2_spawn scripts finished successfully, however, I never get command line access to the multi-vehicle simulation as before. Also, the SITL_UDP_PORT numbers seem correct (14560/14562) even though the ROS_MASTER_URI is a different port number.

[image]

I try to connect via Dronekit in Python but I get a wait_ready timeout which signals that Dronekit did not get parameters in time. I did some debugging and found the simulation to be getting stuck at this line.
[image]

Specifically, it is line 756 which polls the UDP port for simulation data. This poll never gets anything positive and thus waits forever. Please help. My team relies on the stable firmware to test our planes. I updated the repo because we want to use RTPS but now our old environment doesn't seem to be working. Please help. I've attached the sdf, rcS, and launch files so anyone can duplicate our environment.

[multi-plane-SITL-files.zip](https://github.com/PX4/Firmware/files/1639868/multi-plane-SITL-files.zip)

mago3421, 2018-01-18:
Just some notes on this, I think the offending issue comes from one of the submodules. I checked out v1.6.5 which would've been older than the last working stable branch I had. However, the repo forces you to update to the latest submodules despite trying _git checkout --recurse-submodules_ in Git 2.15.1.

As a temp fix, I am using an old repo I had on a thumb drive. It works on native Ubuntu but on my VM, it forces me to check out the latest submodules. I'm looking at how to remedy this in the meantime.

dagar, 2018-01-18:
Check the sitl_gazebo version. It changed quite a bit.

lamping7, 2018-01-19:
@mago3421 using today's master (https://github.com/PX4/Firmware/commit/bda929bebb9f158ea7c52bb70d73043e64cda017) I can get my multi-plane sim to work w/o issues. 

Can you launch one vehicle? How about an iris? When I've had the issue you describe, it is because I forgot to source something.

Paste the output of (not a picture of the output please) after you launch:
`rostopic echo -n1 /diagnostics`
and
`rosrun mavros checkid`

mago3421, 2018-01-19:
@dagar Okay, I'll check that version on a backed-up version of the Firmware. It seems likely it would be that submodule as opposed to NuttX, a module, etc.

@lamping7 Yes, I can launch a single plane, a single iris but multi-iris throws an error. I will paste the output tomorrow.

Thank you both for helping.

lamping7, 2018-01-19:
@mago3421 This seems to be an interfacing issue rather than a PX4 or submodule issue, but I'm sure you're not the only one so, let's figure this out. Let me be more specific on the commands above in case you're not familiar with them. 

Can you launch your second plane model by itself?

You'll need to include the namespace for the `checkid` command and execute it for each. You should see something like:
```
$ rosrun mavros checkid __ns:=/uav1
OK. I got messages from 1:1.

---
Received 6605 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 1, 2, 245, 230, 102, 105, 74, 76, 141, 77, 111, 241, 140, 147, 30, 22, 0, 44, 62, 31
Exception in thread Thread-5 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner

$ rosrun mavros checkid __ns:=/uav2
OK. I got messages from 2:1.

---
Received 6284 messages, from 1 addresses
sys:comp   list of messages
  2:1     32, 0, 2, 245, 230, 102, 1, 105, 74, 140, 141, 77, 76, 241, 147, 62, 111, 30, 31
Exception in thread Thread-4 (most likely raised during interpreter shutdown):
```

For the other command, you'll need to grab two messages (one for each instance), so the `n1` becomes `n2` -- `rostopic echo -n2 /diagnostics` This is going to be long, so paste the output to a gist or pastebin.

mago3421, 2018-01-19:
@lamping7 Much thanks Anthony! I thought that re-declaration of SITL_UDP_PRT would be benign if left in the launch file. This was a relic after getting multi-iris SITL to work last year. Will try and report back so we can hopefully close the bug.

lamping7, 2018-01-20:
The iris model is different in that the model is generated with xacro. 

That argument in the launch file is used to populate the value here:
https://github.com/PX4/sitl_gazebo/blob/5e3375facdfae687c4eb3d20aa9da65aea0ec5d1/models/rotors_description/urdf/iris_base.xacro#L9
which then goes here:
https://github.com/PX4/sitl_gazebo/blob/5e3375facdfae687c4eb3d20aa9da65aea0ec5d1/models/rotors_description/urdf/component_snippets.xacro#L145

It does nothing for sdf models and needs to be added in manually.

mago3421, 2018-01-31:
@lamping7 Just an update, I cloned the stable branch today and copied my sdf, rcS, and launch files that were posted here. I deleted the mavlink_udp_port arguments and ensured that each plane sdf file had the mavlink_interface plugin configured per your specification. The sdf file did have the mavlink_interface plugins configured like above except the robotNamespace and imuSubTopic lines preceded the MAVLink lines. I tried it both way and still could not get the command line on the multi-vehicle SITL. I'm going to try on my home machine and report back.

lamping7, 2018-02-01:
The order of the elements shouldn't matter. If this still isn't working, maybe you need to start with a clean environment and start over to rule out setup issues. Maybe re-read this old issue thread -- https://github.com/PX4/Firmware/issues/6408

closed 2018-06-25 by lamping7