PX4/PX4-Autopilot #8713: tfmini not working on V4 hardware

ChristophTobler, 2018-01-17:
https://github.com/PX4/Firmware/pull/8548 doesn't work for me on a Pixracer or Intel Aero...
I have tried on a Pixhawk (v3 target) and there it worked. On Pixracer and Intel Aero, I get
```
ERROR [tfmini] immediate read failed
ERROR [tfmini] Task exited with errno=11
```
Any Ideas @ayushgaud?

I was disabling mavlink on those ports https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L545-L554

dagar, 2018-01-17:
Which serial port is it connected to? https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/tfmini/tfmini.cpp#L84

ayushgaud, 2018-01-17:
> Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L545-L554

I think it could be because of the hard coded serial port address. You may
try modifying it according to your platform.

ChristophTobler, 2018-01-17:
Sorry, forget to mention that I actually did that...
I used `tfmini start /dev/ttyS3` on the Intel Aero (which should be the telemetry port) and `tfmini start /dev/ttyS1` on the Pixracer (TELEM1, if I remember correctly)

dagar, 2018-01-17:
@ChristophTobler did you stop mavlink on those interfaces (or prevent it from starting)?

ChristophTobler, 2018-01-18:
@dagar I stopped the instances mentioned in the first comment. That should be enough right?
Can I stop a stream on a specific device (e.g. `/dev/ttyS1`) without stopping mavlink completely (`mavlink stop`)?

ChristophTobler, 2018-01-18:
fix in https://github.com/PX4/Firmware/pull/8726.

closed 2018-01-29 by ChristophTobler