PX4/PX4-Autopilot #8715: MS5525 Sensor does not compensate properly.

tubeme, 2018-01-17:
@dagar 

Hi Daniel, 

I have some flights with MS5525 enabled and the graph does not look good at all. We are currently flying without the airspeed enabled, but the sensor is connected and the readings are there.

The sensor is calibrated once on the field and we left it like this. S don't be suprised that the sensr is not zeroed.

You can clearly see the problem arround 10 deg celsius. I remember 10 degrees and 30 degrees were limit values for the compensation mechanism. So they dont work properly.

https://logs.px4.io/plot_app?log=09a5175f-634a-49f6-a18b-49d22bbab4ca

https://logs.px4.io/plot_app?log=cf8767bf-3933-417a-8fca-e807db8223dd

I have more logs from the past 6 days so if needed I can upload more.

priseborough, 2018-01-17:
A ground test is required to eliminate some possibilities. Please enable pre-arm logging, disconnect all tubes to the sensor and  do a ground test where it starts outdoors below 10 deg C and is then brought indoors. Also try another sensor of the same model if you have one.

dagar, 2018-01-17:
Log 1
[image]

Log 2
[image]

dagar, 2018-01-17:
I know other people using these successfully in the winter. I'll see if I can get logs to compare.

priseborough, 2018-01-17:
The possibilities of blocked tubes creating a temperature dependant pressure differential or a faulty sensor need to be eliminated.

dagar, 2018-01-17:
Here's the airspeed and temperature from another setup with an MS5525 just this week.

[image]

tubeme, 2018-01-18:
@priseborough We use brand new pitot from 3DR. The sensor is mRo. The pitot is clean and never crashed. We changed it with the mRo original one, but I'm pretty sure it is not the pitot. 

During the MS5525 driver creation by @dagar We made about 10 bench tests involving the freezer and hair drier to simulate the temp changes. I have them posted in the MS5525 PR back then. And I remember that we had exactly this problem around 10C and 30C.

If we calibrate bellow 10C and fly bellow 10C then no problem. (Your case @dagar temp was about -2C)) If you calibrate and fly between 10-30C again no problem and if you calibrate over 30C and fly over 30C again no problem. 

The problem occurs when you calibrate let say at 20C and hen fly bellow 10C or above 30C.

Here are two more flights that you can see the compensation is not OK. Notice what is the reading of the sensor before flight and after the flight. Conditions are the same.

Calibrated on the field:
https://review.px4.io/plot_app?log=b624b49b-c35a-4021-a78b-875fe736b63b

Not calibrated on the field.
https://review.px4.io/plot_app?log=3261e46d-eeb4-4ef8-9184-e7f6d0eb4d03

I will report with the change of Pitot and if no change we will change the sensor with a new one as well. I will create some bench tests with the freezer.

tubeme, 2018-01-18:
It happens that in California we fly very often around these 10C so we see this problem exaggerated. 

There is 10-15 m/s difference between the start of the flight and the end of it. If we flew in airspeed mode you know what are the consequences...

dagar, 2018-01-18:
In your logs I would suspect something physical with the pitot/tubing, or perhaps a problem with the sensor itself. I had 2 MS5525s for development and one of them was broken. It still provided readings, but they were wrong.

We also shouldn't assume a single problem. Going back to the original PR that added support for the MS5525 I believe there was a mix of pitot issues, early driver bugs, and lack of understanding about the sensor itself.

One thing that can be confusing for bench testing is the behaviour during extreme temperature changes (eg take it out of the freezer). The air temperature used to calculate the corrected differential pressure changes immediately, but the temperature of the sensor itself lags behind.

tubeme, 2018-01-18:
@dagar I will agree with you for now until we cite hardware problem. The thing is that the set of sensors that I bench tested are different set than the one we use for the flights... 

We changed the Pitot with the original one. We made a bench mark LOG with calibration at 15C. While flying around 10C will give us pretty decent info. And we have a new sensor ready to double-check if the sensor itself is the problem.

Thanks

closed 2019-02-18 by stale[bot]