PX4/PX4-Autopilot #8717: Invalid command: rc_receiver

Seanmatthews, 2018-01-17:
Using:
* PX4 1.7.3
* Snapdragon Flight w/latest firmware
* Qualcomm UART ESC
* Spektrum DX8 w/recommended receiver

I upgraded from PX4 1.6.5 to 1.7.3, using the same px4.config and mainapp.conf as before. I build PX4 using the command `make eagle_legacy_default` as I did with version 1.6.5. Upon starting PX4, the ESC lights blink to show the board has initialized. However, PX4 complains with `Invalid command: rc_receiver` and, as expected, I cannot manually control the vehicle.

ChristophTobler, 2018-01-18:
It got changed in https://github.com/PX4/Firmware/pull/7798. You have to swap `rc_receiver` for `spektrum_rc` in your px4.config

Seanmatthews, 2018-01-19:
@ChristophTobler `spektrum_rc start -D /dev/tty-1` still does not seem to let the Spektrum controller connect. Switching back to 1.6.5 and using `rc_receiver` still works fine.

ChristophTobler, 2018-01-19:
It's a lower case d -> `spektrum_rc start -d /dev/tty-1`. I have it on tty-3 (default) and it works

Seanmatthews, 2018-01-22:
Thanks @ChristophTobler , that solves it.

closed 2018-01-22 by Seanmatthews