PX4/PX4-Autopilot #8718: Intel Aero: uncontrollable attitude

santiago3dr, 2018-01-17:
Running current master 850678b and had a flip upon take-off
looking at the log, pitch and roll do not look to be registering:
https://logs.px4.io/plot_app?log=71f9eb76-8ccc-4444-9f6f-5d4b7ab676d1

reverted to yesterday's master 0e65753 and stabilize is fully responsive:
https://logs.px4.io/plot_app?log=072aada3-d5a4-4f28-bb5d-9cea084674b1
Something might have been damaged in the flight controller because once altitude mode was engaged, the copter shot up in altitude and hit the roof.

last successful flight was on 0e65753
https://review.px4.io/plot_app?log=a3026aee-25ab-431a-b076-941084c0cc28

Stifael, 2018-01-18:
hm, this https://logs.px4.io/plot_app?log=072aada3-d5a4-4f28-bb5d-9cea084674b1 looks very strange. there are no velocity setpoints in z direction until about 1 second after you switched to altitude. The should be setpoints in z available as soon altitude mode is engaged. In addition, it looks more like the vehicle looses altitude than gaining once switched to altitude mode.

Stifael, 2018-01-18:
By the way: the successful flight that you mentioned happened outdoors, but the crash with same commit happened indoors. Hence, the comparison is not really fair. It could be very well the case that indoor flight is broken...

jlecoeur, 2018-01-18:
@Stifael the recent changes made to the mixer do not affect `quad_x`, so I doubt that is the cause. Looking at the log it seems the issue is on the sensor or estimation side. As you saw the attitude estimates are not logged, there is also a large variation of magnetic field after take-off.

priseborough, 2018-01-18:
The problem with height control in https://logs.px4.io/plot_app?log=072aada3-d5a4-4f28-bb5d-9cea084674b1 is caused by excessive high frequency vibration in flight that causes a vertical inertial navigation offset that the EKF is unable to correct. Without GPS vertical velocity the EKF is more sensitive to inertial sensing errors.

**Z acceleration**

[image]

**vertical innovations**

[image]

santiago3dr, 2018-01-23:
The main issue was a lack of attitude control running 850678b from the first log mentioned

This log was an indoor test to verify the issue; the first 2 times it flipped in the same manner:
https://review.px4.io/plot_app?log=ca97e8ca-7922-4f1e-89af-8c5a08da1c01

https://review.px4.io/plot_app?log=137c18b8-5948-4670-97b8-4b554f88b6c9

closed 2018-08-01 by santiago3dr