PX4/PX4-Autopilot #8727: Recent changes to master may have caused issues with HITL

arwagoner, 2018-01-18:
As you can see, the sensor set up is fine and everything works as expected when testing with fw v1.7.3. 
[image]

However, when testing with the current master, it says that sensor setup is required and it appears that the sensor messages are not being received. This was tested with an mRo Pixhawk flight controller and a Pixhawk 3 Pro. 
[image]

dagar, 2018-01-18:
I think it was broken in a rebase. Can you try this? https://github.com/PX4/Firmware/pull/8728

arwagoner, 2018-01-18:
That seems to have fixed it. Everything is working as expected again. Thanks! 
[image]

LorenzMeier, 2018-01-18:
I'm closing this. @dagar looks like we need to audit other parts as well to make sure.

closed 2018-01-18 by LorenzMeier

1MSG1, 2018-01-29:
All - It appears that the fix that's in the present master version still leaves X-Plane HIL inoperable for similar reasons as originally tagged. "Unable to Start Mission: vehicle not ready."

I performed a test with the exact same environment using the last known stable version (for me at least) which was 1.5.5 and there were no issues.

dagar, 2018-01-29:
@1MSG1 please open a new issue or ping me on slack later this week so we can get to the bottom of this. I'd like to properly restore X-Plane HIL and keep it working this time.