PX4/PX4-Autopilot #8729: HITL review

dagar, 2018-01-18:
- what currently works
   - jmavsim
   - gazebo
   - xplane
   - flightgear
 - what do we want to support
 - can we make commander oblivious to HIL (commander rewrite)
 - should we consider unifying with the SITL interface (flash considerations aside)

TSC21, 2018-01-18:
@arwagoner maybe you can leave here also your latest experience with the Gazebo side and the current state

arwagoner, 2018-01-19:
In my experience, only quadcopters are reliably supported in Gazebo. The standard vtol airframe is able to takeoff in Gazebo, but it doesn't seem to respond to any landing commands. Neither plane airframe (sys_autostart 1000, 2100) works with Gazebo at all. If it is feasible, it would be great to have all types of airframes (at least their generic versions) working with one simulator.

hamishwillee, 2018-01-31:
Re "what currently works", the docs list is here: https://dev.px4.io/en/simulation/hitl.html#compatible_airframe

- There is no documentation of flightgear, so I'd argue that even if it doesn't currently work even if if in theory it might.
- The [xplane images for settings](https://dev.px4.io/en/simulation/hitl.html#using-x-plane-fixed-wing-only) are for an older version. I got this kind of flying specifying a "Cessna model". It wasn't clear to me whether this actually represents the PX4 UAV or not. Upshot, I'd say "not clear if actually works".
- There is no VTOL HITL support.

I don't know what we want to support, but I'd argue that if we're going to support something:
- It should be supported, tested, and known to work - not "ambiguously so" like Flightgear and Xplane
- We should support at least one of each frame type (vtol, plane, quadcopter) in at least one simulator. 
- The setup in QGC should be the same in all cases.

I like the idea of having a simulator that supports all vehicle types, and gazebo would seem to be the one with the greatest traction.

jingego, 2018-03-06:
I am doing research on HITL platform of Tailsitter. I'm very interested on this function. Could I join this?

hamishwillee, 2018-03-06:
@jingego Sure, but probably better to hold actual "discussion" on slack in #hitl channel.

hamishwillee, 2018-11-14:
@dagar Ping. There has been a lot of chatter on HITL recently. Can we progress this or redefine it in any way to an acheivable goal? Minimally, I'd like to see a HITL setup in automatic testing for Gazebo on copter, plane, vtol.

julianoes, 2019-05-02:
@stale keep it open!

hamishwillee, 2025-03-27:
It sounds like at least for the SITL side of things we're settling on Gazebo - Jmavsim, Flight gear etc are all community supported (effectively deprecated), and Gazebo Classic likely to die when there is parity and we fix our supported Ubuntu version to something that doesn't run it.

No idea bout the HITL, SIH story. But I think we should probably close this, and next time around I'm planning to remove the documentation tag because its just clutter.