PX4/PX4-Autopilot #8730: VTOL: ATTENTION!!! DANGER

tubeme, 2018-01-18:
PUSHER MOTOR STARTS DURING AIRSPEED CALIBRATION
on the bench.

The plane quad delta wing, the tubes from the pitot were swapped.

We recreated it two times

The second time by any chance we had logging enabled all the time so we have it on log.

https://logs.px4.io/plot_app?log=de51dc5e-e868-426f-90c3-1fcc42855fe1

It happened after we blew in the pitot at the end of the calibration. The calibration went wrong (because of the reversed tubes). Both times the motor started.

After we corrected the tubes never happened again.

dagar, 2018-01-19:
I don't see current in your log, but you can tell from the gyro where it started.

[image]

dagar, 2018-01-19:
[image]

tubeme, 2018-01-19:
There is no actuator controll!

But there are 1100us on the servo bus... It is Idle as RPM but still could hurt you. (Out motor is at main ch7)

[image]

dagar, 2018-01-19:
Can you run `pwm info` and share the output?

tubeme, 2018-01-19:
Will try I'm not at the location...

dagar, 2018-01-19:
Questions
1. why did output 6 go to 1100us?
2. why didn't the px4io keep output 6 disarmed?

@tubeme it possible please capture `pwm info` in a few different states. After a fresh boot, armed in MC mode, armed in FW mode.

sanderux, 2018-01-19:
@tubeme why does your log only contain 30% of the params?

tubeme, 2018-01-19:
@sanderux Really don't know...

sanderux, 2018-01-21:
Can you send the param file for your airframe?

tubeme, 2018-01-31:
@sanderux Sorry I was away for a while.

I will get the param file when the person comes back from vacation.

Other then this now the plane does not make this problem. May it was because of the reversed tubes from the pitot... We will try to recreate again.

RomanBapst, 2018-02-13:
@tubeme Did the person come back from his vacation?

tubeme, 2018-02-22:
Yes here is the parameter dump
[Parameter VTL3X 2018_Feb_6 (1.6.5).zip](https://github.com/PX4/Firmware/files/1748151/Parameter.VTL3X.2018_Feb_6.1.6.5.zip)

tubeme, 2018-02-22:
It continues to do it all the time. After calibration for example, every time. It switches the pusher to 1100us.

RomanBapst, 2018-03-07:
@tubeme I was unable to unzip the file you attached. Can you verify that it's a real zip file?

SinigerVTL, 2018-03-07:
Hmmm I use rarlab but I told it to make zip. Probably something went wrong. Can you try a RAR instead...

BTW since we reported this issue, we can reproduce it each and everytime with AUAV X2 + 1.6.5 after calibration. Does not matter what kind of calibration. It does it to the mag cal to the gyro cal or acc cal.

The worst thing we have to disconnect the battery to stop it. The arm button does not affect it.

tubeme, 2018-03-07:
Hmmm I use rarlab but I told it to make zip. Probably something went wrong. Can you try a RAR instead...

BTW since we reported this issue, we can reproduce it each and everytime with AUAV X2 + 1.6.5 after calibration. Does not matter what kind of calibration. It does it to the mag cal to the gyro cal or acc cal.

The worst thing we have to disconnect the battery to stop it. The arm button does not affect it.

tubeme, 2018-03-14:
@RomanBapst Did you manage to open the file or do you need me to re zip it?

closed 2019-02-18 by stale[bot]

hamishwillee, 2019-02-18:
@tubeme @sanderux Should this still be alive?

sanderux, 2019-02-18:
No, never came back with relevant data. Should open a new issue if it can be repeated.