PX4/PX4-Autopilot #8733: VTOL: Front transition Quad Ramp Down problem and Attitude Control.

tubeme, 2018-01-19:
Hi guys,

We have two problems with the quad copter ramp down that are connected.

It is seen from all of our flights without airpseed sensor, despite the mode the quad copter starts to ramp down the thrust since the middle of the transition.

[image]

We think this is not a good way to make the ramp down. From our bench mark system we found that in reality the ramp down is done by the attitude controller of the quad. In full manual we held the throttle at hover level to the end of the transition, and then sharply cut it. So the KK attitude controller automatically lowers the throttle if it feels lift and it achieves the ramp down before the end of the transition. This makes the transition more solid in all conditions.

So the second question is why do we need lift after the half of the transition?

And this is the second issue. Usually we reach the minimum airpseed, ie. the speed at which there is lift in the wing at the end of the transition. Up to this moment there is no lift in the wings and elevons such that to allow the airplane to have a proper attitude control. So the only way to properly implement attitude control in this under speed condition with technically stalled airplane is to make it with the quadcopter. But since we make a very long quad copter ramp down from the middle of the transition there is no way for attitude control from the quadcopter.

So here you can clearly see how our plane has the tendency to pitch up. Up to the middle of the transition the quad womehow holds the pitch up.

[image]

There is no attitude control from the quad copter plus a ramp down:

[image]

And you can see that the only attitude control comes from the airplane, which elevons are stalled at this moment.

We think that atittude control should come from the quadcopter and airplane at the same time during the transitions - front and back. They does not fight each other. And the quadcopter should have hover throttle during the whole transitions back and front.

We could not find a way to control the ramp down, probably in airspeed mode it is different dependent on the speed.

here is the log
https://review.px4.io/plot_app?log=3261e46d-eeb4-4ef8-9184-e7f6d0eb4d03

tubeme, 2018-01-31:
@sanderux  Any ideas?

sanderux, 2018-02-03:
> It is seen from all of our flights without airpseed sensor, despite the mode the quad copter starts to ramp down the vertical thrust from the middle of the transition. We think this is not a good way to make the ramp down.

We do not do this because at higher speeds the quad motors interfere and create drag. you likely want this because you have set your front transition to 5 seconds, you should use 10 minimum.

closed 2018-02-03 by sanderux