PX4/PX4-Autopilot #8739: "Camera triggering" in simulation (SIL/HIL) from PX4

kjetilkjeka, 2018-01-21:
**Feature Request**

When doing simulations of surveying missions, triggering cameras in the simulation is required. I would like to implement this feature in `PX4`.

I've already hacked something together by sending the mav link `COMMAND_LONG` and checking it in AirSim. Since I'm not certain that this is the suitable MavLink message, I also created [this issue](https://github.com/mavlink/mavlink/issues/828)

If i understand the PX4 codebase correctly then the "correct" way to implement it is subscribing to the `camera_trigger` orb in [mavlink simulator](https://github.com/PX4/Firmware/blob/6227139df11cc28939c6a1ccafaa69f5bebe811f/src/modules/simulator/simulator_mavlink.cpp) and send the "correct" mavlink message on the HIL interface when this occurs.

I also hope to [add support for this in AirSim](https://github.com/Microsoft/AirSim/issues/734)

bkueng, 2018-01-29:
Yes, the mavlink simulator should send the appropriate mavlink messages. Similar to this: https://github.com/PX4/Firmware/blob/6227139df11cc28939c6a1ccafaa69f5bebe811f/src/modules/mavlink/mavlink_messages.cpp#L1494. I'm not familiar with all the camera-related mavlink messages, but MAV_CMD_IMAGE_START_CAPTURE seems appropriate.

hamishwillee, 2018-01-30:
@kjetilkjeka This was answered(is) by Lorenz. I am attempting to clarify final detail in https://github.com/PX4/Firmware/pull/8775#issuecomment-361476205 and will then update the docs. Are you OK for this particular instance to be closed?

Just as an aside, my understanding in that comment is as per @bkueng 's comment. We use mavlink to forward any camera messages like MAV_CMD_IMAGE_START_CAPTURE  to the simulator which can then act on them as desired. The camera trigger message will also be forwarded but these are typically only used for matching timestamps between logs and images taken.

kjetilkjeka, 2018-01-30:
@hamishwillee I've gotten my answer on how to do this, feel free to close.

But perhaps this issue, the PR or a new issue/PR related to the documentation should be kept open to track the addition of this in the documentation?

hamishwillee, 2018-01-30:
OK. I have created https://github.com/PX4/Devguide/issues/442 to track the docs. Thank you!

closed 2018-01-30 by hamishwillee