PX4/PX4-Autopilot #8742: Motors not responding

Deanatius, 2018-01-22:
Hello there,
I am facing an irregular problem with my Pixhawk. I have been using the flight controller for about 2 months now and I am using an RC transmitter manually control the drone.

The current flight stack I am using is 1.7.3 stable release.

Everything was working normally last week and I left my quadcopter completely untouched for the weekend. Coming back to it today and I found out that the motors no longer respond. I am able to arm the drone using the transmitter and through QGroundControl but the motors do not idle spin nor does it respond when increasing the throttle. Even the usual twitching does not occur when I connect the battery. I have checked that the Pixhawk, PDB, and RC receiver are turned on as usual.

I used APM Planner to flash arducopter firmware and did all the calibrations and the motors are working perfectly. However, when I flashed back the PX4 firmware through QGroundControl the problem still persists. I did all the calibrations including ESC and got the usual music tone.

LorenzMeier, 2018-01-22:
Could you please share a log file like suggested in the text template when creating an issue? Without the log file we can just guess, which will not get us closer to a solution. Thanks!

Deanatius, 2018-01-22:
Sure thing, here it is.

[log_1_UnknownDate.px4log.tar.gz](https://github.com/PX4/Firmware/files/1651053/log_1_UnknownDate.px4log.tar.gz)

tops4u, 2018-01-22:
X-Post. Just to note that this is also discussed here: http://discuss.px4.io/t/motors-suddenly-not-responding/5489 

@Deanatius is there a special reason to use the old Log Format?

Deanatius, 2018-01-22:
> @Deanatius is there a special reason to use the old Log Format?

Its the format given when I download from QGroundControl

tops4u, 2018-01-22:
When you reinstalled PX4 from scratch it would default to the newer "ulog" Format. Check the param SYS_LOGGER (should be 1). 

Could you Post what ESC you are using? What PWM Signal Type are you using (regular, OneShot)? What are your PWM Settings for the MainOut PWM Connectors? If your ESC are configurable did you change the Settings?

Deanatius, 2018-01-22:
Using a PDB-XT60 with a Turnigy MultiStar 30A BLHeli-S  2~4S (OPTO) connected to each motor.

Using regular PWM signal type with settings:
PWM_MAX = 2000us
PWM_MIN = 1000us
PWM_RATE = 400Hz

nanthony21, 2018-01-28:
I'm seeing this same issue on a pixracer.

closed 2019-02-18 by stale[bot]